arduino/Arduino #3554: Command line doesn't build and leave hex file (OSX)

ajfisher, 2015-07-20:
I'm trying to build an automation script and use the command line to build a series of hex files (for different target boards). If I use 1.6.6 from within the application and do a build then the appropriate hex file will land in the location given at the end of the build process, eg:

`/var/folders/bd/369p_cm53tsb7yzzdmppbnw80000gn/T/build9219197947028832118.tmp/sketch/`

I can cd to that folder and see the hex file available (as well as all the object files etc too).

If I run from the command line to build this (with the same board parameters):

``` shell
~/Downloads/Arduino.app/Contents/MacOS/Arduino node_pixel_firmata.ino --board arduino:avr:uno --verbose --verify
```

Then the output location: `/var/folders/bd/369p_cm53tsb7yzzdmppbnw80000gn/T/build5020081654195010543.tmp/sketch/` in this case is entirely empty - not even any build files.

What's weird is that doing an upload will seem to generate the files but again at completion they are removed. Indeed if you do an upload you can watch all the files be created and then removed as the process finishes in another window.

So the question is, how do I get the command line to retain the files?

cmaglie, 2015-07-20:
This is the expected behaviour. If you want to keep the .hex file you must specify a custom build path by setting the preference `build.path` using the `--pref` syntax:

`--pref build.path=../path/to/my/build/folder ~/Downloads/Arduino.app/Contents/MacOS/Arduino node_pixel_firmata.ino --board arduino:avr:uno --verbose --verify`

the documentation is here:
https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc#preferences

ffissore, 2015-09-08:
You can also pass `--preserve-temp-files`: when passe, temp files won't get deleted

closed 2015-09-08 by ffissore