arduino/Arduino #3556: MEGA_SOFT_SPI set as 1 leads to failure in compiling (SD-library, IDE 1.6.x)

feranick, 2015-07-20:
When: 
#define MEGA_SOFT_SPI 1
in libraries/SD/utilities/SD2Card.h (in order to use software pins for SPI with MEGA boards), IDE fails to compile (error message below). This only happens for IDE 1.6.x. When used with 1.0.5, compilation proceeds as intended and the MEGA is using software pins for SPI. This makes the library useless with 1.6.x, and only 1.0.5 can be used.

This happens on any version of IDE (up to 1.6.5_r3) on OS X 10.10.4

/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp: In member function 'void Sd2Card::chipSelectHigh()':
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp:167:5: error: 'SPI' was not declared in this scope
     SPI.endTransaction();
     ^
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp: In member function 'void Sd2Card::chipSelectLow()':
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp:176:5: error: 'SPI' was not declared in this scope
     SPI.beginTransaction(settings);
     ^
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp:176:26: error: 'settings' was not declared in this scope
     SPI.beginTransaction(settings);
                          ^
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp: In member function 'uint8_t Sd2Card::init(uint8_t, uint8_t)':
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp:275:3: error: 'SPI' was not declared in this scope
   SPI.beginTransaction(settings);
   ^
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp:275:24: error: 'settings' was not declared in this scope
   SPI.beginTransaction(settings);
                        ^
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp: In member function 'uint8_t Sd2Card::setSckRate(uint8_t)':
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp:518:14: error: 'settings' was not declared in this scope
     case 0:  settings = SPISettings(25000000, MSBFIRST, SPI_MODE0); break;
              ^
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp:518:57: error: 'SPI_MODE0' was not declared in this scope
     case 0:  settings = SPISettings(25000000, MSBFIRST, SPI_MODE0); break;
                                                         ^
/Applications/Arduino.app/Contents/Java/libraries/SD/src/utility/Sd2Card.cpp:518:66: error: 'SPISettings' was not declared in this scope
     case 0:  settings = SPISettings(25000000, MSBFIRST, SPI_MODE0); break;
                                                                  ^
Error compiling.

facchinm, 2015-07-21:
Hi @feranick , you also need to comment out `#define USE_SPI_LIB` from `Sd2Card.h` and  `Sd2Card.cpp`. Probably cpp define is a bug so thank you for the report.
I'll wait for your feedback before closing the issue

feranick, 2015-07-21:
Thanks, Martino. Commenting the string fixed it. 
Thanks again for the fast reply.
Nicola

closed 2015-07-22 by facchinm

MagisterLudiFRZ, 2017-07-25:
I was faced with the same problem and I comment out #define USE_SPI_LIB from both Sd2Card.h and Sd2Card.cpp. But IDE fails to compile again.

```
Arduino: 1.8.3 (Windows 8), Board:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\libraries\SD\src\utility\Sd2Card.cpp: In member function 'uint8_t Sd2Card::setSpiClock(uint32_t)':

C:\Program Files (x86)\Arduino\libraries\SD\src\utility\Sd2Card.cpp:546:3: error: 'settings' was not declared in this scope

   settings = SPISettings(clock, MSBFIRST, SPI_MODE0);

   ^

C:\Program Files (x86)\Arduino\libraries\SD\src\utility\Sd2Card.cpp:546:43: error: 'SPI_MODE0' was not declared in this scope

   settings = SPISettings(clock, MSBFIRST, SPI_MODE0);

                                           ^

C:\Program Files (x86)\Arduino\libraries\SD\src\utility\Sd2Card.cpp:546:52: error: 'SPISettings' was not declared in this scope

   settings = SPISettings(clock, MSBFIRST, SPI_MODE0);

                                                    ^

exit status 1
Error compiling for Arduino/Genuino Mega or Mega 2560.
```

Zhigalin, 2019-02-24:
S

> Same here. Compile fails on setSpiClock.

Same here

Zhigalin, 2019-02-24:
Solution:  
( from https://forum.arduino.cc/index.php?topic=487918.0 )  
Move the compiler directive "#define USE_SPI_LIB" down to the line right after "#ifndef SOFTWARE_SPI"   in your Sd2Card.h file (in my case found under "C:\Program Files (x86)\Arduino\libraries\SD\src\utility\")  
In your Sd2Card.cpp file, wrap the line "#define USE_SPI_LIB" with a compiler directive so that it is not   included if SOFTWARE_SPI is defined -- it should look like this:  

```
#ifndef SOFTWARE_SPI  //Added to enable use of MEGA
 #define USE_SPI_LIB   
#endif  //SOFTWARE_SPI
```

DON'T FORGET to move the #include "Sd2Card.h" statement so that it is above the modification you made in the previous step, otherwise the SOFTWARE_SPI flag will not have been set and USE_SPI_LIB  will get defined... so now it should look like this:
```
#include "Sd2Card.h"
#ifndef SOFTWARE_SPI  //Added to enable use of MEGA
  #define USE_SPI_LIB   
#endif  //SOFTWARE_SPI
```
Edit the SD.cpp file (C:\Program Files (x86)\Arduino\libraries\SD\src\) to modify the SDClass::begin method with compiler directive to exclude the line "card.setSpiClock(clock) &&" when we are using SOFTWARE_SPI mode -- mine looks like this:
```
#ifndef SOFTWARE_SPI    //Added compiler directive so that MEGA works with this code
  boolean SDClass::begin(uint32_t clock, uint8_t csPin) {
    return card.init(SPI_HALF_SPEED, csPin) &&
           card.setSpiClock(clock) &&
           volume.init(card) &&
           root.openRoot(volume);
  }
#else
  boolean SDClass::begin(uint32_t clock, uint8_t csPin) {
    return card.init(SPI_HALF_SPEED, csPin) &&
           volume.init(card) &&
           root.openRoot(volume);
  }
#endif  //SOFTWARE_SPI
```
That's it!  That should get things working.