arduino/Arduino #3564: Alternate SPI configuration with SPI transaction function

jbhome, 2015-07-22:
This issue relates to use of the spi transaction function with respect to device libraries that handle SPI configuration internally.
While one could dig into the library code for each device to determine how it configures the SPI device, it would seem better, (to me anyway) to have a configuration read-back function.
One would instantiate each device and let it configure SPI, Then use a SPI library function to read-back and save the current configuration. Even better, integrate this into the transaction function giving it the ability to read/save/use the current configuration. So the final process would be instantiate each device , use the transaction function to read/save the SPI configuration for each device.
The result is the same as the current transaction process but seems to better encapsulate/manage   library based SPI configuration.

Thanks for the opportunity to discuss this.

Regards,
Jim Brooks

PaulStoffregen, 2015-07-22:
Which libraries?

jbhome, 2015-07-31:
Ether_card and u8G.
Each devices uses SPI but with different clock/phase configurations.
I thought more about and IMHO using the spi.transaction function within the library would seem to work best.

Regards,
Jim Brooks