arduino/Arduino #3565: IDE 1.6.3 does not work with old Freeduino Diecimila (file upload blocks)

VogonJeltz, 2015-07-22:
IDE 1.6.x does not work with old Freeduino Diecimila.
Long time ago, I once had it running with IDE 0.9 and 1.05, requiring 
-  FTDI-USB-driver FT232R (install by CDM206.exe)
-  FT232R adjust by MPROG.EXE.

Now it doesn't work any more although virtual COM port has been installed correctly and is detected by Windows (X?32).

But file upload is not possible any more via USB.

I now get the errors

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xee
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xe6
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x62
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xbd
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x29
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x09
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x09
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x09
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x09
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x09

The same program (a LED1602 test sketch from Arduino playground ) uploaded to a Mega 2560 works fine, no issues.

Is this a IDE 1.6.3 bug about the Freeduino Diecimila boards?
Is this fixed perhaps in 1.6.5?
or what might be the reason?

VogonJeltz, 2015-07-27:
update:
patching boards.txt  (sync speed->19200 and 328.p ->328 ) didn't resolve the problem but just shifts it to different additional libs.

VogonJeltz, 2015-07-27:
as the issue has not been resolved, please re-open it again so that other user may see this issue and participate in a solution.

VogonJeltz, 2015-07-28:
no one here to resolve this issue and make my boards work with the IDE 1.6 again like before with the 1.0.5 ?

matthijskooijman, 2015-07-29:
If you can specifically state what needs changing in the IDE, or provide a pullrequest, we would be happy to fix things. However, debugging all kinds of third-party hardware is tricky and too time consuming, so we recommend you contact the manufacturer of your Freeduino for support, or try the Arduino forums where there might be other users willing to invest some time to help you out.

closed 2015-07-29 by matthijskooijman

VogonJeltz, 2015-07-29:
no, the question is perhaps simply:
what has been changed from IDE 1.0.5 to 1.6.3 so that with the 1.0.5 the Freeduino  worked and for the 1.6.3 it don't work any longer?

And the request is: please change it back !

(no idea what a pullrequest is and how to issue that!?!? )

VogonJeltz, 2015-07-29:
it is unreasonable that you close this topic unless my questions have been answered!

please re-open this issue!

PaulStoffregen, 2015-07-29:
No, the unreasonable thing here is your expectation for Arduino to support and resolve technical problems with a third-party non-Arduino product.

VogonJeltz, 2015-07-30:
why??
Arduino is open source and public domain!

Chris--A, 2015-07-30:
I feel your frustration however the Arduino mob cannot be held responsible for third party's products.
Freeduino should ensure the stability of their products by issuing a fix (which is quite easy, as the IDE is open source as you pointed out).

> patching boards.txt (sync speed->19200 and 328.p ->328 ) didn't resolve the problem but just shifts it to different additional libs.

What do you mean by this?

I can see in the history that the baud rate was changed from 19200 to 57600.
Can you try upgrading your bootloader?

VogonJeltz, 2015-07-30:
I have patched the boards.txt for Duemilanove from 57600 back to 19200, 
but without success.
I also have re-newed the FTDI-USB-driver as written above, also without sucess.

What do you mean about the bootloader? I haven't changed anything about the bootloader,  It worked before, why shouldn't it work any longer?
As I have 2 of the Freeduinos and no one workes any longer it's not a single issue but a general issue about the Freeduinos.

matthijskooijman, 2015-07-30:
It seems you are not entirely understanding how the Arduino project is
organized, so let me try and clarify some things. Note that I'm not part
of the Arduino team, so this should not be seen as an official view,
though I'm pretty sure the Arduino team has the same view.

> Arduino is open source and public domain!

AFAIU the term "Public Domain" means that no copyright applies to some work,
or that it has expired. Copyright applies to pretty much all things
related to Arduino (board designs, core code, IDE code, website, logo),
so none of those are in the public domain.

Most of the Arduino stuff is open-source, yes. This means that the
copyright holders (mostly the developers from Arduino LLC, and any
contributors in the community) grant you, and everyone else, a license
to use, modify and distribute the software and hardware designs
(possibly subject to restrictions). It does not automatically mean that
any user of the software automatically gets to decide what the software
should or should not be doing. It certainly does not give the users the
right to decide how the developers should be spending their time.

Arduino as a whole is a community, but Arduino LLC is of course still a
company. They make money selling hardware, but have made their software
available under an open source license, free of charge. The license
allows you to take the code, modify it in any way you see fit. You can
build your own IDE using the code as well (though you cannot use the
Arduino name for it, due to trademark limitations). However, the
Arduino IDE is maintained by the Arduino team, and they get to decide
what they code and features they include. In general, input from the
Arduino community is appreciated, especially if someone has gone through
the trouble of figuring out in detail how something should work, be
implemented or even supplies the code for it. Still, even if all the work
has been done by the community, the Arduino team is still entitled to
decide that the code is of insufficient quality, or that a feature does
not fit with their plan for *their* Arduino IDE (which I think is their
stance on most printf suggestions).

Additionally, the Arduino team has gone to great lengths to allow using
the Arduino IDE with third-party hardware, like your Freeduino. This is
a great service to the community, and more than users should expect from
Arduino (given that Arduino LLC has no direct benefit from third parties
selling more hardware). In general, the Arduino team is also willing to
change the code or fix bugs that affect third-party hardware if there is
a clear picture of what needs to change (or better, a pull request), but
you cannot expect the Arduino team to help all IDE users with debugging
their third-party hardware. After all, using the IDE does not make you a
Arduino LLC customer. Buying official Arduino hardware does, but buying
a Freeduino makes you a Freeduino customer, that should contact
Freeduino for support.

> If Arduino changes anything and Hardware I purchased some months ago
> does not work any longer I expect to make it work again like before!

You seem to argue that, because your board used to work with 1.0.x and
now breaks with 1.6.x, it's Arduino's fault for breaking it, and they
should make sure things get fixed again. However, it is unreasonable to
expect the Arduino IDE team to actively test and support all the
hundreds of third-party boards that are out there, that claim to be
Arduino compatible.

Furthermore, perhaps you do not realize what the purpose of this issue
tracker is? It is used by the Arduino team to keep track of things that
need their attention or work from them. This includes bugs in the IDE,
feature requests that they think are useful, etc. It might make sense to
simply allow issues for all things even somehow related to Arduino,
allowing the community to respond to them even if the Arduino team does
not have any time or priority, but in practice this will just mean that
many of these issues will just remain unanswered and open indefinitely,
which completely removes the utility of the issue tracker to get an
overview of things that need to work on (with the large number of issues
opened daily, it is already tricky to keep it useful). To get help from
the community, the forums are better suited anyway, since a lot more
users read them.

> (no idea what a pullrequest is and how to issue that!?!? )

In short, a pullrequest is a way to present changes to the code in a
convenient and easy to review way. Try Google or github docs for more
info on how this stuff works.

I hope that my explanations help you to understand better how things
work as they do, since then the time I invested writing this will have
been worthwile. Nobody is trying to harass you or closing your issues
just to annoy you, but you have been making some unreasonable requests.
At some point, this also means that people (myself, for one) are tired
of investing time into the questions you ask, or the suggestions you
give and you might get shorter responses, or none at all in the future.

(Note that some of my terminology might be technically incorrect given the
recent Arduino LLC vs Arduino SRL conflict, but the end result for you
and other users should not change)

PaulStoffregen, 2015-07-30:
Indeed, you need to direct your frustration and anger at whoever made & sold that Freeduino board.  Arduino is a very open platform.  They are the ones who are responsible for making sure their hardware works with the Arduino software.  Testing non-Arduino brand boards is not Arduino's responsibility.

Many of us do make non-Arduino brand boards and support them well.  Next time you buy (or board) a board, consider choosing one that has better support.  If you want one that Arduino LLC actually supports, get a real Arduino brand board!

VogonJeltz, 2015-07-30:
when I started with Arduino 1 year ago I purchased this Arduino starter set (actually 2 of them for 2 different working places) by Conrad and Franzis not knowing that there are several brands and companies. As it was called Arduino starter set to me it seems it was just Arduino (what else?).
Now of course I'm disappointed because the money I once had spent for the stuff which once was working fine is now suddenly squandered useless and I now can throw it in the garbage.