arduino/Arduino #3566: Linux install package should contain script to create Dash item.

Avamander, 2015-07-22:
It would be much nicer if the Linux package also contained script to add Arduino IDE to the Dash menu.

mschlenker, 2015-07-24:
I admit this. Upon first start a file `$HOME/.local/share/applications/arduino-1.6.5.desktop` might be created as a simple softlink to the not yet existing `arduino-1.6.5/arduino.desktop`. When updating the IDE and deleting old versions orphaned softlinks to old IDEs do not point to valid desktop files anymore and thus are ignored.

Should I change the startup script accordingly and add a desktop file?

Avamander, 2015-07-24:
Maybe an additional script for those who do have the privileges, everything will continue to work the same way, but those who wish can get a shortcut to dash. (Also, why isn't Arduino package updated in the Ubuntu repos?)

Avamander, 2016-03-13:
@ffissore I guess you aren't working on this :) ?

NicoHood, 2016-03-13:
Isnt this fixed now?

ffissore, 2016-03-14:
Indeed I'm not. See
https://groups.google.com/a/arduino.cc/forum/#!msg/developers/YJLX6AZHem4/kpsLef7XBAAJ

Avamander ha scritto il 13/03/2016 alle 20:00:

closed 2020-06-08 by Avamander