arduino/Arduino #3567: AVR: sprintf() not working for floats

VogonJeltz, 2015-07-22:
for AVRs sprintf() is not working for floats (on Due it works though). 
To make it work, a lib.c float patch has to be issued.
this is actually unreasonable.
Please make this feature possible by #include or anything.
On very small boards one may refrain from #including it to save memory nevertheless.

per1234, 2015-07-23:
You can use [dtostrf()](http://www.nongnu.org/avr-libc/user-manual/group__avr__stdlib.html#ga060c998e77fb5fc0d3168b3ce8771d42) instead on AVRs. It's an avr-libc function so you need to `#include <avr/dtostrf.h>` to use it with non-AVR boards.

VogonJeltz, 2015-07-23:
no never!!
I need sprintf() for multi-format and multiple-parameter arguments! Anything else is nonsense:

sprintf(sbuf, "poll: %03d - heading: %6.2f  pitch: %5.2f roll: %5.2\n", dvar, fvar1, fvar2, fvar3);
Serial.print(sbuf);

matthijskooijman, 2015-07-23:
Due to the way sprintf works, enabling the float support is only possible by adding a linker option, there is not any reasonable way I can think of to enable it from inside a sketch using an include or whatever.

There has been some discussion on the list about (s)printf recently, and it was clear that it is not part of the supported Arduino API, so no extensive effort will be made to more easily support it.

If you really need float support, you will need to modify the compiler options by modifying platform.txt or boards.txt. Ideally, you could add additional compiler options per sketch to simplify this, but that isn't supported yet (IIRC there is an issue about this already).

I'm closing this issue as wontfix. If you have specific ideas about *how* to fix this, feel free to leave another comment (or submit a pullrequest).

closed 2015-07-23 by matthijskooijman

VogonJeltz, 2015-07-23:
as by copying the lib.c float patch it seems there is a way to fix this issue for AVRs and make sprintf process floats by AVRs.

So would it be possible to have 2 different sprintf versions, e.g.

sprintftiny() just for small AVRs w/o float support
sprintf() like for ARM/Due feat. float support
?

matthijskooijman, 2015-07-23:
I'm not sure what this patch is. Do you have a link?

AFAIK, avr-libc provides two sprintf libraries, both providing a `sprintf` function, one with float support and one without, and you can pass a flag to the linker to make it use the float-enabled version instead of the regular, float-less version.

As for the editor - use triple backticks ``` before and after a code block. I have updated your comment to apply this.

VogonJeltz, 2015-07-23:
to the lib.c float patch I have made an upload in our forum, please look here if you wish:
http://www.mindstormsforum.de/viewtopic.php?f=70&p=67477#p67238
http://www.mindstormsforum.de/download/file.php?id=742
I'm not sure if you must be registered for download, please tell me if it doesn't work for you!

can you please make this topic "open" again" ?

matthijskooijman, 2015-07-24:
> to the lib.c float patch I have made an upload in our forum, please look here if you wish:

Hm, I can download those, but the zip only contains a compiled libc version, without source code or an actual .patch file, so that doesn't really tell me anything. I suspect that this might be a version of avr-libc that has float support enabled by default, removing the need for a  custom linker option.

I had another look at the [linker options required](http://www.nongnu.org/avr-libc/user-manual/group__avr__stdio.html#gaa3b98c0d17b35642c0f3e4649092b9f1) and the [printf sources](http://svn.savannah.nongnu.org/viewvc/trunk/avr-libc/libc/stdio/vfprintf.c?revision=2191&root=avr-libc&view=markup) in the hope that there might be some way to automatically including the float-supported printf by referencing a special symbol, but it seems that each version exports exactly the same symbols, so that won't work.

> can you please make this topic "open" again" ?

Since I'm still not sure what Arduino should be changing to improve this and it's unlikely that a lot of effort will be invested in printf-related stuff, I'd rather keep it closed until it's clear that something needs changing.

One thing I could imagine being changed is to make it easier to supply custom compiler options from a sketch, but that would warrant a separate issue (which I think already exists).

> How can I issue a pull request?

That's a bit of a broad question, please check out the github docs and other stuff written about this in Google.

VogonJeltz, 2015-07-24:
yes, the lib contains ALMOST a complete lib.c version. It must be copy-and-pasted into the program folder to replace the original ones.

Well, about this sprintf issue:
I would like to get this resolved so:
I WOULD APPRECIATE VERY MUCH YOU TO RE-OPEN IT AGAIN !

Maybe other readers will have ideas about how to fix or resolve this issue.
The way it's  currently -not-  working is unacceptable:
The ANSI C standard has to be performed as a default (float-suppport).
Deviations from the ANSI C standard must be initiated by the user (no float support).

Thank you!