arduino/Arduino #3568: [Feature request] Abort uploading

NicoHood, 2015-07-22:
There needs to be an abort uploading button. I always have the problem that avrdude cant connect with my arduino for some reason (of developing) and then I have to kill the task. This happens if you want to upload to a mega and an uno is still selected for example.

matthijskooijman, 2015-07-22:
Since currently, pressing the upload button twice also causes problem, switching the button between "upload" and "abort upload" is probably good. The same goes for verify, btw.

Avamander, 2015-07-24:
Or if your computer suddenly decided to remap the connected devices when one connected arduino decides to draw too much current, you start the upload to wrong arduino then aborting upload can save an Arduino.

NicoHood, 2015-07-24:
I also want to mention that on Linux on several circumstances the avrdude process is still alive after an error message. The IDE will continue working but open another avrdude thread.
After a while you can hear that avr dude is still open, because my laptop fan goes crazy. The single CPU task avrdude takes 100% for a core. I have to manually kill them. Maybe I should flag this as separate issue?

NicoHood, 2015-08-16:
Please also add this to the burning bootloader option. This freezes the IDE and is really annoying.

ffissore, 2015-09-08:
Merged into #3771

closed 2015-09-08 by ffissore

lmihalkovic, 2015-12-04:
[LMCHK]