arduino/Arduino #3570: b=b++;   It depends on the version of software

pjordi, 2015-07-22:
This program with the version 1.0.5, b increases the value by 1, and with the version 1.6.5, b=0 all time.

int b=0;
void setup() {
  Serial.begin(9600);
}

void loop() {
  b=b++;
  Serial.println(b);
  delay(100);        
}

carlosperate, 2015-07-22:
`b=0` from 1.6.5 is the correct result, `b++` should return the value before the increment takes place.

bobc, 2015-07-22:
This is a classic C/C++ "gotcha".

b=b++ is "undefined behaviour". http://stackoverflow.com/questions/4968854/is-i-i-truly-a-undefined-behavior.

Undefined behaviour means the result depends on the compiler, as you have found, and both 1.0.5 and 1.6.5 are producing correct results, according to the standard.

You should change your code - to increment b, use one of "b++;" "b = b + 1;" or "b +=1".

carlosperate, 2015-07-22:
Wow, thanks @bobc , learning something new every day.

Chris--A, 2015-07-23:
It appears a large amount of reversal is occurring due to optimizations. Your case has popped up many times, however I have recently seen ordering differences in the way the compiler evaluates function parameters and even allocation of global addresses (https://github.com/arduino/Arduino/issues/3061), which are implementation defined.

The key is to avoid assumptions. If you're unsure, stack overflow can be a big help, however the latest (free) standard can be accessed here : http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2013/n3797.pdf

facchinm, 2015-07-23:
Thanks everyone for the clear explanation!

closed 2015-07-23 by facchinm