arduino/Arduino #3572: 1.6.5 crashes when switching sketches

matthijskooijman, 2015-07-23:
On IRC, someone reported a problem with the IDE locking up when changing sketches through the File -> Sketchbook menu. The user does not really speak English and doesn't have a github account, so I'm relaying this here. Perhaps someone else can reproduce this (I tried on Linux but failed, the user reported it breaking on OSX and Win7 in the same way).

Basically when switching sketches, every now and then the interface would lock up. He wasn't sure if the lockup was complete, but at least clicking the "upload" button would not do anything at all (not even the "clicked" feedback of changing the button color). I've asked him to reproduce and see if e.g. typing code still works. Killing the IDE and restarting it would allow uploads again.

No error was shown on screen, I've asked him to start from a terminal to see if any errors try again.

He will also try 1.6.4 to see if the bug is also present there.

matthijskooijman, 2015-07-23:
A few more observations:
 - Nothing is printed on the terminal
 - When it crashes, the interface full freezes, no code can be typed anymore either. Even more, when you click a sketch in the menu, it freezes *before* the new window with the new sketch is opened, so the old window becomes unusable.

matthijskooijman, 2015-07-23:
After leaving the crashed IDE on for a while, this message is shown in
the bottom panel:
```
Exception in thread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" Exception in
thread "JmDNS(fe80:0:0:0:0:0:0:1%lo0.local.).Timer" java.lang.OutOfMemoryError: GC overhead limit exceeded
  at javax.jmdns.impl.DNSCache.allValues(DNSCache.java:348)
  at javax.jmdns.impl.JmDNSImpl.cleanCache(JmDNSImpl.java:1770)
  at javax.jmdns.impl.tasks.RecordReaper.run(RecordReaper.java:58)
```

Chris--A, 2015-07-23:
I have no problems on windows 8.1

This may be of interest, a few others are having trouble with compiling on windows XP: https://forum.arduino.cc/index.php?topic=337541.0

matthijskooijman, 2015-07-23:
Not sure if the linked forum thread is related, they talk about crashing while compiling, which sounds significantly different from the freezes in this report.

I wonder if there is something in the OP's environment that causes this. Any ideas?

Original reporter reports that this also happens with 1.6.4. Using that, he got this  backtrace:

```
Exception in thread "AWT-EventQueue-0" java.lang.OutOfMemoryError: GC overhead limit exceeded
	at java.lang.Class.newReflectionData(Class.java:2511)
	at java.lang.Class.reflectionData(Class.java:2503)
	at java.lang.Class.getInterfaces(Class.java:845)
	at sun.awt.SunToolkit.isInstanceOf(SunToolkit.java:2046)
	at sun.awt.SunToolkit.isInstanceOf(SunToolkit.java:2051)
	at sun.awt.SunToolkit.isInstanceOf(SunToolkit.java:2051)
	at sun.awt.SunToolkit.isInstanceOf(SunToolkit.java:2051)
	at sun.awt.SunToolkit.isInstanceOf(SunToolkit.java:2051)
	at sun.awt.SunToolkit.isInstanceOf(SunToolkit.java:2051)
	at sun.awt.SunToolkit.isInstanceOf(SunToolkit.java:2036)
	at javax.swing.LookAndFeel.installProperty(LookAndFeel.java:275)
	at com.apple.laf.AquaTextFieldUI.installListeners(AquaTextFieldUI.java:53)
	at javax.swing.plaf.basic.BasicTextUI.installUI(BasicTextUI.java:804)
	at javax.swing.JComponent.setUI(JComponent.java:666)
	at javax.swing.text.JTextComponent.setUI(JTextComponent.java:328)
	at javax.swing.text.JTextComponent.updateUI(JTextComponent.java:338)
	at javax.swing.text.JTextComponent.<init>(JTextComponent.java:312)
	at javax.swing.JTextField.<init>(JTextField.java:231)
	at javax.swing.JTextField.<init>(JTextField.java:172)
	at edu.mit.blocks.codeblockutil.LabelWidget$BlockLabelTextField.<init>(LabelWidget.java:407)
	at edu.mit.blocks.codeblockutil.LabelWidget$BlockLabelTextField.<init>(LabelWidget.java:383)
	at edu.mit.blocks.codeblockutil.LabelWidget.<init>(LabelWidget.java:50)
	at edu.mit.blocks.renderable.BlockLabel$1.<init>(BlockLabel.java:81)
	at edu.mit.blocks.renderable.BlockLabel.<init>(BlockLabel.java:81)
	at edu.mit.blocks.renderable.PageLabel.<init>(PageLabel.java:10)
	at edu.mit.blocks.renderable.RenderableBlock.<init>(RenderableBlock.java:245)
	at edu.mit.blocks.renderable.RenderableBlock.<init>(RenderableBlock.java:186)
	at edu.mit.blocks.renderable.FactoryRenderableBlock.<init>(FactoryRenderableBlock.java:40)
	at edu.mit.blocks.workspace.PageDrawerLoadingUtils.loadBlockDrawerSets(PageDrawerLoadingUtils.java:236)
	at edu.mit.blocks.workspace.Workspace.loadWorkspaceFrom(Workspace.java:849)
	at edu.mit.blocks.controller.WorkspaceController.loadFreshWorkspace(WorkspaceController.java:397)
	at com.ardublock.core.Context.resetWorksapce(Context.java:110)
```

ffissore, 2015-08-04:
The error is caused by ardublock, as the last line (error originator) says 
`    at com.ardublock.core.Context.resetWorksapce(Context.java:110)`

ffissore, 2015-08-04:
Well, let me rephrase it.
It's a memory error: the IDE has finished its available RAM.
The last piece of code that required RAM and failed to allocate it was the discovery thread (first stacktrace) and ardublock (second stacktrace). This does not mean that either discovery or ardublock are broken: it means they were unlucky and asking for RAM when it was over.

But, AFAIU, ardublock is an IDE tool. Tools/extensions *integrate* with the IDE, and such integration is troublesome at times.

User should first try uninstalling ardublock, use a *vanilla* IDE and see if the problem persist. If so, then this issue can be valid and further investigation should be conducted. Otherwise, it's invalid and it should be reported to ardublock

matthijskooijman, 2015-08-04:
@ffissore, I'll keep an eye out to see if this users comes back on IRC and ask him.

matthijskooijman, 2015-08-13:
The original reporter tried without Ardublock and that worked, so it's an issue there. It seems Ardublock has a known memory issue(https://github.com/taweili/ardublock/issues/77) , which can be solved by increasing the Java heap size  AFAICS. Would it make sense to increase that size by default in the IDE? I'm not sure if that causes any side effects, I would expect it is just the maximum size? 

Also, perhaps the IDE should somehow handle out-of-memory issues more gracefully? Not sure what would be needed for that, though...

ffissore, 2015-08-14:
I've attempted to increase max mem to 1G this February, but for some users it was too much. See #2749. Out of memory errors are hard to handle. Ideally you should remove the cause of the leak but it's bit like stopping a ship from sinking using a cork: you may reach the leak too late and your cork may be insufficient. I'll look for some strategy

closed 2015-08-14 by ffissore