arduino/Arduino #3575: Library Manager inclusion request

ReefPOM, 2015-07-23:
I have created a library, and believe it is formatted v1.5 appropriately for the Library Manager.   bitbucket.org/ReefPOM/ospom

cmaglie, 2015-07-27:
@ReefPOM I'm sorry be we do not support bitbucket now, because the library manager use the Github's CDN to serve the .zip archives of the libraries.

I'm sorry to say that you must move your repo to github (or maybe just copy it if you want to keep it on bitbucket).

ReefPOM, 2015-07-27:
Thank you,  I have moved it to GitHub:    https://github.com/ReefPOM/OSPOM

Once included, if I push version tag updates, they will be automatically updated on arduino.com (with a delay)?

Chris--A, 2015-07-28:
The libraries are scanned once a day I'm pretty sure, so at the most 24 hours.

cmaglie, 2015-07-29:
Done, thank you!

> The libraries are scanned once a day I'm pretty sure, so at the most 24 hours.

the libraries scan is done every hour so, even better, you'll get updates published within an hour.

closed 2015-07-29 by cmaglie

ReefPOM, 2015-07-30:
When should I expect to see it on 'playground.arduino.cc/Main/LibraryList'?  I don't mind waiting, I'm just new at this, and am thinking maybe I didn't do something correctly.

facchinm, 2015-07-31:
Hi @ReefPOM , your library has been listed in the library manager (inside the IDE), not on the website; that page is in Playground so you can update it manually including your lib if you wish :smile: 

[image]

ReefPOM, 2015-07-31:
So Cool!  Thank you :-)