arduino/Arduino #3576: Sketch Crashes MCU

CreateiveRobitics, 2015-07-23:
Hi, I found a problem with a compiled sketch that looked like an issue either with libraries, or the compiler/tool-chain because it was causing very mundane code to break and crash the MCU

I'm using IDE release version 1.6.5 and the target is a Sparkfun pro-mini 328 5v 16Mhz
I was using the ServoTimer2 lib and the PID lib but didn't hit any problems until I wrote the function below.

The problem occurred when I added this function:
void waitForServo(){
while(1){
    if(!HipServo.isMoving()) return;
    updateSystem();
    delay(1);
  }
}
HipServo.isMoving() just returns 1 or 0 and updateSystem() calls a series of system functions for checking the serial port and updating a servo motion system and the PIC library - ALL of these functions have worked fine until this point. Calling this new function will cause the sketch to print garbage over the serial port and then run the sketch setup() function - Is this a soft reset? -- some global variables were not being reset and neither were some peripherals, for example a PWM would continue to run. Also, **This was not a brownout!** there is no problem with power on my system, the problem occurs when all other components are disconnected and I just run a bare board off of the FTDI USB adapter.

When I added some serial prints to the function as follows:
void waitForServo(){
  int n = 0;
  Serial.print("Starting: ");
  while(1){
    if(!HipServo.isMoving()) return;
    Serial.print("Waiting: ");
    Serial.println(n);
    n++;
    updateSystem();
    delay(1);
  }
}
The function was being called repeatedly and n was not incrementing. - The call to this function is not inside a loop and should only be called once. If I removed the return statement, making the while loop impossible to escape from, it would STILL call the function repeatedly and never increment n.

I'm sorry I can't actually post the whole sketch because I managed to fix the issue in the end. The sketch now works so I can't replicate the problem. 

My updateSystem() function was as follows:
void updateSystem(){
  if( (millis()-lastUpdateTime) < updateInterval) return;
  //This all takes 3ms to execute
  
  refreshSerial();
  Motors.refreshSensors();
  Sensor.refresh();
  HipServo.refresh();
  hipPosition = HipServo.getPosition();
  updateBalancePID();
  pidSetMotors();

  lastUpdateTime = millis();
}

I discovered that moving refreshSerial() into the sketches loop() function fixed the issue, so I then rearranged the code that was called by refreshSerial() and moved the function call back, and now it works **My serial port code had been working fine with no issues**  .... I figured that the behaviour was so bizarre it must be a compiler or tool-chain issue, or maybe a buffer overflow/memory problem from a library (The serial or stream libraries!?), and that restructuring some other bits of my code might shift the profile of the compiled code enough to get over the issue - it looks like it was!

The complete sketch included three c++ classes I wrote myself, none of which do anything naughty with memory or pointers. Sketch uses 14,942 bytes of program storage space. Global variables use 831 bytes of dynamic memory.

I realise this is a bit vague as bug reports go due to the lack of code and reproducibility but even though my code now works it was unsettling to see what looks like a bug serious enough to crash the MCU - Just to head off any 'are you sure it wasn't a beginners mistake' - yes, very sure, I'm a computer science PhD ...   but then again ...

Chris--A, 2015-07-24:
If your sketch is restarting, you are most probably clobbering memory. Either the stack and heap have clashed, or possibly you accessed a bad memory location, either directly or by over-running an array.

Making use of the WDT `MCUSR` should give you a reason why the system restarted.

This is however, a question for the forums: https://forum.arduino.cc

cmaglie, 2015-07-24:
@CreateiveRobotics 
You should file a report only when you find exactly where a bug is, or when you have a simple test case that can be reproduced. From your report I can only deduce that **maybe** there is bug **somewhere** in the Arudino code, this is really too vague...

Sometimes we consider to analyse also bigger sketch if the problem is relevant, but in this case the code of the Serial of the atmega328p is probably one of the most used part of Arduino and it's really well tested so, for now, until you have something to work on I'm closing the issue.

> I realise this is a bit vague as bug reports go due to the lack of code and reproducibility but even though my code now works it was unsettling to see what looks like a bug serious enough to crash [...]

Even experts sometimes fails to spot a bug and blame the compiler: 99.9% of the times are user's bug and not compiler's bug. Have you already tried to enable all warnings from the preferences?

closed 2015-07-24 by cmaglie

CreateiveRobotics, 2015-07-24:
My mistake was writing it up at the end of a long and frustrating day instead of sleeping on it ... I realise now there is a chance it is a memory problem and how breaking up a function could have fixed it. I also remembered this morning that all my code is in dropbox so I can recover the non working version. I had already ruled out any buffer overflow or bad memory access in my own code.

I will note that I have had several inexplicable problems when using the Serial library, with sketches hanging and seeming to depend on how many print statements I included (too few being as much as an issue as too many ...) ... I also actively avoid the String object because it had, in the past, caused serious problems with any code I wrote - even dirt simple sketches - and all the info I could find from the forums indicated that there was a bug in the way memory allocation worked with String... I didn't look into those claims too deeply though.