arduino/Arduino #3580: TTY Port Problems for IDE v > 1.6.1

acnebs, 2015-07-24:
System: MBPr on OS X Yosemite, 10.10.4

I started out using version 1.6.5 of the Arduino IDE to upload programs to my Arduino Uno. Unfortunately, in the port menu, no tty.* ports were showing up - only cu.*. I am able to upload a sketch via the /dev/cu.usb* port, but only once (I'm assuming this is just the nature of the cu device). Also serial communication is impossible via the cu port. The TTY devices existed in /dev/, so I knew it wasn't my machine. I swapped out the cable and tried my other Arduino. No luck.

Then I tried downgrading to version 1.6.1 of the IDE. Specifically, I tried the version labeled "Java 7+". The TTY ports show up fine, and I am able to upload consequent sketches without unplugging/plugging back in the USB cable.

Then I tried 1.6.3. Same problem as 1.6.5. CU ports show up (Bluetooth and USB), but no TTY.
I will probably try 1.6.2 next to see if the problem exists in that version. Will update this ticket with any such info.

So in conclusion: there is a problem with the Arduino IDE for versions above 1.6.1 that prevents TTY ports from showing up on Mac OS X Yosemite.

acnebs, 2015-07-24:
Update: 1.6.2 does not work either.

Last working version (in regards to displaying TTY ports) on Mac OS X Yosemite is 1.6.1.

ffissore, 2015-08-04:
Are you using a genuine Arduino? Otherwise it could be a driver issue, unrelated to the IDE

closed 2015-09-08 by ffissore