arduino/Arduino #3581: can't remove folder durably since once installed a zip lib from there

VogonJeltz, 2015-07-24:
I can't remove folder durably since once installed a zip lib from there by mistake.
It was a temporary project folder containing also this zip lib  (by mistake saved to this folder) and then added as a zip lib by the lib manager.
As I encountered thsi mistake, I couldn't remove it from the list of installed libs, and since I simply removed this temp folder it now always rises from the dead any time I restart the IDE.
it containes just 1 readme.txt file, saying
"Informationen zur Installation von Libraries finden sich unter: http://arduino.cc/en/Guide/Libraries"

so I have 2 questions:
How can I remove this permanently?
How can one uninstall libs in the IDE in general?

VogonJeltz, 2015-07-28:
no one here  to fix this 1.6 IDE issue?

facchinm, 2015-07-29:
The libraries you install via zip file method are saved in the `sketchbook/libraries` folder. 

Under Windows, it will likely be called `My Documents\Arduino\libraries`. For Mac users, it will likely be called `Documents/Arduino/libraries`. On Linux, it will be the `libraries` folder in your sketchbook.

Delete the library you installed there (by deleting its folder) and you are done. 
The Library manager has no method to delete a library as a security measure.

VogonJeltz, 2015-07-29:
no, I can delete the directory (for me it's d:akten\programmierung\arduino\libraries) ,
but as I already explained: when I restart the Arduino IDE this effing ...\libraies folder again resurrects !
I can't delete it permanently, once and for all!

ffissore, 2015-08-04:
You can't. The folder must be there. An empty `libraries` folder has NO effect on the IDE behaviour

closed 2015-08-04 by ffissore

VogonJeltz, 2015-08-04:
Please read more closely befor you answer!!

I DON'T WANT THIS EFFING LIBRARIES FOLDER IN MY WORKING DIRECTORY, ESPECIALLY IF IT'S EMPTY !!
I WANT TO REMOVE THIS CRAP FOLDER PERMANENTLY!!

matthijskooijman, 2015-08-05:
This is probably going to help anything, but I'll say it anyway: You've now totally crossed my trolling-tolerance for being rude, demanding and non-constructive. I'll no longer invest any time in helping you out. I can imagine others feeling the same, so if your questions and remarks remain unanswered, this is probably why.