arduino/Arduino #3582: PulseInLong not documented

matthijskooijman, 2015-07-24:
A while ago, `pulseInLong()` was added, but it doesn't seem documented yet. Also, I think `pulseIn()` was slightly changed, so that documentation should probably be updated too (perhaps a single page for both, since they mostly work the same?).

See 
https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/cores/arduino/wiring_pulse.c#L63
and
https://github.com/arduino/Arduino/pull/2495

agdl, 2015-09-03:
@matthijskooijman take a look please!

https://www.arduino.cc/en/Reference.PulseInLong
https://www.arduino.cc/en/Reference.PulseIn

matthijskooijman, 2015-09-03:
Seems better, thanks. One thing that I think is incorrect is the timeout parameter: looking at the implementation, the timeout is a total timeout, so the function returns 0 if no *complete* pulse was received within the timeout (so even if the pulse starts, but not finishes within the timeout). Also, it might be good to clarify that if the pin is already high when the function is called, it will wait for the pin to go low and then high before it starts counting (which is somewhat contained in " waits for the pin to go HIGH", but it might be good to further clarify).

agdl, 2015-09-03:
@matthijskooijman done!

matthijskooijman, 2015-09-03:
What did you change exactly? I can only find a remark in pulseInLong at the timeout parameter, but a few other places still mention the incorrect behavior. It's probably good to clarify that the timeout starts when the function is called, not when the pulse starts.

agdl, 2015-09-04:
What I changed:

Destription:
* Please also note that if the pin is already high when the function is called, it will wait for the pin to go LOW and then HIGH before it starts counting.*.

Parameters:
*timeout (optional): the number of microseconds to wait for the pulse to be completed: the function returns 0 if no complete pulse was received within the timeout. Default is one second (unsigned long). *

Returns:
*the length of the pulse (in microseconds) or 0 if no pulse is completed before the timeout (unsigned long) *

matthijskooijman, 2015-09-05:
Ok, but what is still incorrect is:

> Gives up and returns 0 if no pulse starts within a specified time out. 

in both pages and:

> returns the length of the pulse (in microseconds) or 0 if no pulse started before the timeout (unsigned long) 

in the pulseIn page.

Also, you didn't add this to the pulsedIn page, only pulseInLong:

> Please also note that if the pin is already high when the function is called, it will wait for the pin to go LOW and then HIGH before it starts counting..

agdl, 2015-09-07:
@matthijskooijman should be ok now!

matthijskooijman, 2015-09-07:
Looks good to me, thanks!

closed 2015-09-07 by matthijskooijman