arduino/Arduino #3584: Building IDE from source

VaclavSal, 2015-07-25:
I am sorry to post this here , but there is no visible way to ask question on 
https://github.com/arduino/Arduino/wiki/Building-Arduino

What is the MINIMAL Java SDK I can use running XP?
And what are the suffixes on Java SDK  names for ?
Do I need Java x SE ?

Chris--A, 2015-07-26:
You now need version 8 of the SDK.

The SE is the one I used I'm pretty sure.

VaclavSal, 2015-07-26:
I get a warning about the "latest Java SDK "  is not guaranteed  (?) to work properly on XP , but it won't even finish Setup! 

Seems silly that Java SDK must have the latest Windows to perform.

Does same limitation apply to Linux?
I rather switch to Linux than buy Windows whatever the latest  and greatest version is.

Chris--A, 2015-07-27:
There are problems that will arise in all operating systems over time (and yes linux does have problems.. baud rates for one). Which is why active development is good. We can find, fix and release.

As for windows XP,
> ## Q: Can I install JDK 8 on Windows XP? Will it work?
A: Yes, as long as you install JDK 8 Update 25 or later. Earlier updates of JDK 8 had an installer issue that prevented installation in Windows XP.

And here is a video with a helpful title: https://www.youtube.com/watch?v=Cfnv41O4voY

PaulStoffregen, 2015-07-27:
I'm pretty sure you'll encounter problems if you try to install Java 8 on a Linux distro from 2001 (when Windows XP was released - three years before Ubuntu started) or even 2008 (when XP SP3 was released).

VaclavSal, 2015-07-27:
Chis,
thanks for the note. I'll have to admit I did not pay much attention which JDK  8 I was installing. 
I'll try it again. 

Can I change subject?

If I decide to start using Linux, what would be the most popular AKA useable , stable flavor to use to work on Arduino.cc  IDE ?
I have a copy of Unbuntu but it is pretty old.

ffissore, 2015-08-03:
I've started using ubuntu since 7 years ago. Other options are available at http://distrowatch.com/dwres.php?resource=major

closed 2015-08-03 by ffissore