arduino/Arduino #3585: how to compile with -F ???

VogonJeltz, 2015-07-25:


VogonJeltz, 2015-07-25:
I get this error msg:
avrdude: Expected signature for ATmega328P is 1E 95 0F
         Double check chip, or use -F to override this check.

But there are no entries to find to use this flag -F or whatever this F-thing is.
Please enable it by a menu entry in the IDE!

matthijskooijman, 2015-07-25:
This sounds like you have selected the wrong board in the IDE, or somehow the programmer or bootloader fails to read the processor signature correctly. If you would add -F (which you can do in platform.txt, if you insist), you would likely run into more errors.

What board are you using? What board have you selected in the IDE? Can you paste the full (verbose) upload output?

VogonJeltz, 2015-07-26:
Actually the boards are od Freeduinos (Diecimila ATmega328) , using a FTDI chip.
1 year ago, when I started with Arduinos from the scratch, I had to install the FTDI drivers, and then it worked (IDE 0.9, 1.05 by Diecimila Board). 
But then I started to use Megas und Dues and the Freeduinos  were sort of forgotten.
Now I reactivated them in order to make a UART-to-I2c converter, But it didn't work no more.
I even installed a new FTDI driver manually from http://www.arexx.com/downloads/CDMv2.10.00__PID_FIX.zip - but no changes.

When I target Diecimila, I get the error as shown above, when I target Unos, then it says the avrdude is out of sync.

So what should I do?

How can I try this -F thing ?

mikaelpatel, 2015-07-26:
I think this line is the problem? https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/boards.txt#L107
If the "...mcu=atmega328p" is given to avrdude it will return a signature error. Change to "atmega328" and test that. 
Cheers!

VogonJeltz, 2015-07-26:
thank you very much, this sounds very promising :)

But what text file is it actually which I have to patch? 
My Arduino program folder is C:\Programme\Arduino\

VogonJeltz, 2015-07-26:
ps,
I think I found it.
But now it still doesn't compile, although targeting the Due it compiles fine::

C:\Programme\Arduino\libraries\UTFTQD\UTFTQD.cpp:56:4: error: #error "Unsupported AVR MCU!"
   #error "Unsupported AVR MCU!"

~~~
#include <SPI.h>
#include <SD.h>
#include <UTFTQD.h>
#include <ardustdio.h>

//=====================================================================================
// misc.  
//=====================================================================================

#define  _DUEMISO_    74  // Arduino Due SPI Header
#define  _DUEMOSI_    75
#define  _DUESCK_     76

#define  clock()      millis()  
#define  LRAND_MAX    32767
#define  srand(seed)  randomSeed(seed)
#define  rand()       random(LRAND_MAX)
#define  rando()      ((float)rand()/(LRAND_MAX+1))

//=====================================================================================
// TFT LCD
//=====================================================================================
#define  UTFT_CSpin    52    // <<<<<<<< adjust!

// set LCD TFT type
int16_t  LCDTYPE   =   -1;

#define  __LCD1602__    1  // LCD1602  Hitachi HD44780 driver <LiquidCrystal.h> 
                           // http://www.arduino.cc/en/Tutorial/LiquidCrystal
#define  __SERLCD__     2  // Sparkfun serLCD 16x2  
                           // http://playground.arduino.cc/Code/SerLCD
#define  __UTFT__       4  // Henning Karlsen UTFT 2.2-2.4" 220x176 - 320x240 lib
                           // http://henningkarlsen.com/electronics/library.php?id=51

//--------------------------------------------------------------------------------------------------
//UTFT   myGLCD(Model, SDA=MISO, SCL, CS,         RESET,  RS)    // Due: 3 exposed SS pins: 4,10,52
//UTFT   myGLCD(QD220A,   A2,    A1,  A5,         A4,     A3);   // adjust model parameter and pins!
  UTFT   myGLCD(QD220A,   50,    49,  UTFT_CSpin,  0,     51);   // A0->Vc (LED), A4->BoardReset
extern   uint8_t SmallFont[];
//--------------------------------------------------------------------------------------------------

#define  lcdWhiteBlack()  {                                                                 \
   if(LCDTYPE==__UTFT__) { myGLCD.setColor(255,255,255); myGLCD.setBackColor(  0,  0,  0);} \
}

#define  lcdNormal()      {                                                                 \
   if(LCDTYPE==__UTFT__) { myGLCD.setColor(255,255,255); myGLCD.setBackColor(  0,  0,  0);} \
}

#define  lcdInvers()      {                                                                 \
   if(LCDTYPE==__UTFT__) { myGLCD.setColor(  0,  0,  0); myGLCD.setBackColor(255,255,255);} \
}

#define  lcdWhiteRed()    {                                                                 \
   if(LCDTYPE==__UTFT__) { myGLCD.setColor(255,255,255); myGLCD.setBackColor(255,  0,  0);} \
}

#define  lcdRedBlack()    {                                                                 \   
   if(LCDTYPE==__UTFT__) { myGLCD.setColor(255,  0,  0); myGLCD.setBackColor(  0,  0,  0);} \
}

#define  lcdYellowBlue()  {                                                                 \     
[... 102 lines cut ...]
   _maxy_ = LCDmaxY / fonthi;   
   memset(wspace, ' ', _maxx_);
   wspace[_maxx_]='\0';
   lcdcls();
   sprintf(sbuf, "LCDtype=%2d %4dx%4d",LCDTYPE,LCDmaxX,LCDmaxY);
   Serial.println(sbuf);
   lcdprintxy(0, 0, sbuf);

}
//=====================================================================================

//=====================================================================================

//=====================================================================================

//=====================================================================================
void loop(){
   char     sbuf[128];
  
   

}
//=====================================================================================
//=====================================================================================
~~~

VogonJeltz, 2015-07-26:
if I change the line bach to 328p, then it says again:

avrdude: Expected signature for ATmega328P is 1E 95 0F

Chris--A, 2015-07-27:
The library would have something like this:

```arduino
#if defined(__AVR_ATmega328P__)
  //...
#else
  #error "Unsupported AVR MCU!"
#endif
```
You could change it to this:

```arduino
#if defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328__)
  //...
#else
  #error "Unsupported AVR MCU!"
#endif
```

VogonJeltz, 2015-07-27:
thanks, I'll try ASAP!
But 3 questions about this issue:

1.)  what about this -F parameter? isn't it possible to drop all this driver patching?
2.) Why is this complicated driver lib patching by 328 vs. 328p necessary?
3.) Couldn't we have an IDE which supports both 328 and 328p chips ?

VogonJeltz, 2015-07-27:
now I also patched the UTFT driver, now I still get
~~~

avrdude: Expected signature for ATmega328 is 1E 95 14
         Double check chip, or use -F to override this check.
Falscher Mikrocontroller gefunden. Ist die richtige Platine im Menü Werkzeuge > Platine ausgewählt?
~~~

Why T** F*** isn't it working any longer? With 1.05 everything was working fine!

facchinm, 2015-07-27:
From the UTFT author's page (http://www.rinkydinkelectronics.com/library.php?id=51) the changelog reports

```
v2.81	21 May 2015	•	fixed a bug which stopped the library from compiling with Arduino 1.6.x
```
I believe you are using outdated libraries (I can't find any reference of "UTFTQD.h" in Google so it must be very old).

Anyway, Diecimila has atmega328p mounted (https://github.com/arduino/Arduino/blob/fe3fa56a1577df9be104df1c91754b9dd4d1a1c2/hardware/arduino/avr/boards.txt#L107) so there's no need to change anything in any file.

Then, I believe that this kind of issues are better tackled on the forum, because they are not related with the IDE in any way.

closed 2015-07-27 by facchinm

VogonJeltz, 2015-07-27:
UTFTQD.h is just Henning Karlsen UTFT.h patched by QD systems.
It's not original Arduino, it's Freeduino.

The UTFT error appears only after first the boards.txt has been patches (sync speed->19200  and 328.p ->328 )

So please re-open it again, it's not a UTFT issue, it's an IDE  issue because the board is not recognized correctly (please read from the start!)

VogonJeltz, 2015-07-27:
and what about this:
1.) what about this -F parameter? isn't it possible to drop all this driver patching using just -F ?
2.) Why is this complicated driver lib patching by 328 vs. 328p necessary?
3.) Couldn't we have an IDE which supports both 328 and 328p chips ?

facchinm, 2015-07-27:
Of course the board isn't recognised because it's not an Arduino board :wink:  
Anyone can support other boards by providing an hardware board file, which will seamlessly integrate with the IDE (even better with the board manager).

The -F flag is DANGEROUS because it could brick your chip if you select the wrong board from the dropdown (and you don't have an ISP programmer)

VogonJeltz, 2015-07-27:
the board has been recognized correctly by IDE 0.9 and by 1.0.5,
and the test code works fie using the Mega 2560 and the Due.
So 
- the code is fine,
- formerly the Freeduino board was fine for 1.0.5 and 0.9
- so why are there issues now about the 1.6.3 ??

facchinm, 2015-07-27:
`avrdude` has been updated in the meanwhile, and that section of the board file has been updated for the last time with this commit (76645b957d8) on `8 Dec 2013`
If you really wish the newer IDE to support your board, add a proper board.txt following this guide (https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#referencing-another-core-variant-or-tool)

VogonJeltz, 2015-07-27:
I have no idea how to do that, I'm just a Arduino newbee (< 1year hobby programming) so I need support how to use my Freeduino boards just  as before by 1.0.5 ... :-/

facchinm, 2015-07-27:
Sorry, did't know about that. I can only encourage you to post your issue on the forum, which is the best place for these kind of issues

VogonJeltz, 2015-07-27:
as the issue has not been resolved, please re-open it again so that other user may see this issue and participate in a solution.