arduino/Arduino #3586: please enhance Serial by a printf() method like ANSI C printf!

VogonJeltz, 2015-07-25:
please enhance Serial by a printf() method like ANSI C printf!
so output by multi-format-multiple-argument list, including e.g.,  %d, %ld, %c, %f, %o, %s, + - 0  .* ...
as shown here:
 http://www.cplusplus.com/reference/cstdio/printf/

matthijskooijman, 2015-07-25:
@cmaglie, I'll leave this one for you...

> I'm really curious why this effing code block formatting does not work! What a poor editor!

You should be using  backticks (same key as the tilde ~), not normal quotes. I corrected them for you.

VogonJeltz, 2015-07-26:
thanks a lot, then it's just about the first part (the printf thing) yet... ;)

q2dg, 2015-07-26:
This has been discussed a lot in Developer mailist: https://groups.google.com/a/arduino.cc/forum/#!searchin/developers/printf/developers/7KpdLDgFsO0/IdW9tSmnCAQJ

VogonJeltz, 2015-07-27:
I don't see there if or how the issue has been fixed meanwhile, but Serial.printf() by multiple parameters and multiple formats is   important to be provided (not just a single "%f" but all of the formatters by a format string!)
Just like, e.g.
~~~
   Serial.printf ("Characters: %c %c \n", 'a', 65);
   Serial.printf ("Decimals: %d %ld\n", 1977, 650000L);
   Serial.printf ("Preceding with blanks: %10d \n", 1977);
   Serial.printf ("Preceding with zeros: %010d \n", 1977);
   Serial.printf ("Some different radices: %d %x %o %#x %#o \n", 100, 100, 100, 100, 100);
   Serial.printf ("floats: %4.2f %+.0e %E \n", 3.1416, 3.1416, 3.1416);
   Serial.printf ("Width trick: %*d \n", 5, 10);
   Serial.printf ("%s \n", "A string");
~~~

pwillard, 2015-07-31:
sprintf() is a nominal workaround

VogonJeltz, 2015-07-31:
theoretically there are many workarounds, but my topic is about the issue, not about a workaround.
Moreover, sprintf() lacks of the "%f" formatter for AVRs, so actually this would not even be a workaround
:-/

PaulStoffregen, 2015-07-31:
I implemented Serial.printf() on Teensy.  If Arduino ever wants to implement this feature, it's only a small amount of open source code.

> it's currently extremely annoying that if one wants to write a line containing multiple strings and multiple numbers one has to write each single string and each single number individually

I agree, it is annoying to *write* several lines, when you know how to do it with a single printf() line.

But Arduino's design decision is about other people *reading* the code, especially those not familiar with the complexities of printf format strings.  For them, *reading* sketches with complex syntax is annoying.  So is having to learn a cryptic syntax before being about to write similar lines.

I imagine you're unmoved by this thinking, but my point it not to convince you.  My point is to explain *why* Arduino has resisted Serial.printf().  Obviously you don't yet understand their point of view, because the best argument you've been able to muster is that you, personally, as an experienced C programmer, are annoyed by having to write multiple simple lines rather than 1 complex line.

Can you at least try to imagine how that must sound to a group of designers who overriding goal is simplicity for a novice-oriented system that priorities ease-of-learning above all else?

q2dg, 2015-07-31:
@PaulStoffregen What do you think about https://github.com/arduino/Arduino/issues/3546 (and https://github.com/arduino/Arduino/pull/3551) ?

PaulStoffregen, 2015-07-31:
I have mixed feelings regarding both of those.

Likewise, I believe printf() is great for experts, but I also recognize the Arduino Team's concerns.

I do not have all the answers to these sorts of tough design trade-offs!

VogonJeltz, 2015-07-31:
printf is not for experts, it's just for C programmers who are used to stdio.h, and it's just the same for sprintf() which is already implemented by Sketch (apart from this effing float formatting issue for AVRs).
The rest it's all the same: 
for printf(), 
for sprintf(), 
for fprintf(), 
for vsprintf(),
for Serial.printf()...

PaulStoffregen, 2015-07-31:
Ok then, I guess that's a "no", you can't try to understand Arduino's point of view?

VogonJeltz, 2015-07-31:
not quite,
if one has a small AVR, one does not need to #include this lib feat. full stdio.h prinf() functionality and having more free RAM instead,
but who wants  full stdio.h prinf() functionality should be able to #include a lib in order to have this functionality

VogonJeltz, 2015-07-31:
I think the point is:
Arduino is coding in C,
and for C there are standards like ANSI C definitions and syntax plus additional standardized libs, e.g.
stdio.h (printf, sprintf, fprintf, gets, fgets, scanf,...)
plus other ones like math.h, stdlib.h,...
First of all, Arduino should apply to the standards,
and if Arduino wants to create additional API funtions to make it easier for beginners or even experts who lack of RAM, it's surely appreciated!
But 1st of all come the (complete) standard features.

PaulStoffregen, 2015-07-31:
For anyone else who wants Serial.printf(), please understand these 2 arguments do **NOT** work:

1) Convenience for experienced C programmers: Arduino is a system designed for novices.  Ease of learning for absolute is the overriding design goal.  Any API requests for advanced features must consider their immediate and long-term (as experts publish examples and libraries) impact on novices.

2) Standards compliance: Arduino achieved success where many others failed, because of its design focus on simplicity, even when in conflict standards and well-established (but complex) programming practices.  Attempting to argue standards compliance, at the expense of simplicity, is utter ignorance of the path that made Arduino so successful.

In closing this issue, I want to give some guidance to those who will inevitably being up Serial.printf() again and again.  Please, do yourself a favor, try to understand Arduino's design goals.  Express your argument from a viewpoint of understanding of Arduino's design goals.

closed 2015-07-31 by PaulStoffregen

VogonJeltz, 2015-07-31:
I am not experienced in C -
I started programming C 1 year ago, when I started programming with Arduinos!
But I have minor experience using Lego Mindstorms NXC.

I understand thzat Arduino implements beginner-friendly API functions, but that does not mean that Arduino has to exclude all basic C facilities such as stdio.h functions additionally.
Finally sprintf() is already there, but admittedly mutilated on AVRs.

The C concept is based on libraries which vcan be optionally #included or not in order to decide which funtionality is wanted and which one not. 
- Ardiono should just apply to this principle.

q2dg, 2015-07-31:
This is another approach: http://www.arduiniana.org/libraries/streaming

VogonJeltz, 2015-07-31:
I don't see whether this has anything to do with my intention of my TOP!
My TOP is about a multi-format plus a multi-argument-list of parameters passed to a serial print function, comparable to printf() writing to stdout (strings, ints, floats, bytes, formatted by %C %s %i %f %E all at once)
I hate it having to write each parameter  one by one !

BrentWilkins, 2015-08-03:
Does this help? http://playground.arduino.cc/Main/Printf

VogonJeltz, 2015-08-04:
thank you, but they say:
"Also note that by default the Arduino IDE does not set the AVR linker options to support floating point in the xxprintf() routines. So while that saves quite a bit of code space on the AVR builds, it means that printf() functions cannot be used for floating point output on the AVR. "
To have the %f or %E formatters is indispensable!