arduino/Arduino #3591: Arduino Due 2nd USB port: useable as a USB host for USB keyboards?

VogonJeltz, 2015-07-27:
hi,
about the Arduino Due's 2nd USB port: 
is it possible to make it useable as a USB host for USB keyboards?

facchinm, 2015-07-27:
Yes, you can use the native usb as a host (not the programming usb)
https://www.arduino.cc/en/Reference/USBHost

closed 2015-07-27 by facchinm

VogonJeltz, 2015-07-27:
thank you,
is this port USB OTG compatible ? (most microUSB-to-USB(A)female adaptors are USB-OTG)

facchinm, 2015-07-27:
OTG is not implemented (so the Due can't switch between host and device roles at runtime). Anyway, the cable you suggest is compatible and the Due will act as a host accepting any device (like mice, USB memory sticks and so on)

VogonJeltz, 2015-07-27:
thank you very much!

VaclavSal, 2015-07-27:
Small note - be aware that the "standard" Arduino USB keyboard class is implemented as "loop" and NOT as a interrupt. Makes usage of USB port as "universal" AKA using the bus for multiple devices a challenge. 
Also  the class is a "fork" ( open source) from commercial code which has better documentation. 

But it does work as keyboard USB device.

VogonJeltz, 2015-07-27:
thank's, I'll see, I just ordered the microUSB-to-USB(A) adapter!

q2dg, 2015-07-27:
In https://www.arduino.cc/en/Reference/USBHost it is said this library is for the Due only. Isn't it for Zero?? Thanks!

VogonJeltz, 2015-07-28:
what?
My topic is about the Due.
I have a Due, I want to plug a keyboard to it's 2nd USB port, and the reference is just about this purpose. 
No one mentioned the Zero - but if you have issues with the Zero, don't mind to open your own topic!

facchinm, 2015-07-29:
@q2dg it's also for Zero of course, thanks for pointing that out. We'll update the Reference ASAP.
Thanks!

VogonJeltz, 2015-08-06:
I now tried it, but unfortunately the keyboard functionality is more than poor:
https://github.com/arduino/Arduino/issues/3630
No standard keyboard  functions are avalable (kbhit, getch, getche, getchar, gets, scanf),
the implementation of the US.task is weird,
keyPressed and keyReleased don't return anything and it's totally osbscured how and when they do either or anything meaningful
 usb.Task blocks program execution intermediately for many seconds.
All the USB keyboard functionality should have been implemented eithe by stdin and the ANSI C sdtio.h functions,
or, alternatatively, as a workaraound,
at least by a task of Scheduler, running independently and continuously, producing a stream by itself from which one could read keyboard input by customized kbhit, getch, getche, getchar, gets, scanf functions.
The way it's implemented for the moment anyway is not only "experimental" as it's written but actually just unsatisfactory, useless and with a gibberish API, not C syntax as one could have expected.