arduino/Arduino #3594: Skip the start screen in one-off mode

rakyll, 2015-07-27:
arduino, as a command line tool, should never require a graphical desktop environment. Given one of the one-off mode options (such as --verify and --upload), the executable should only use stdout and stderr to report the launch progress or keep itself silent.

carlosperate, 2015-07-27:
A known issue that for now it looks like it won't be fixed: https://github.com/arduino/Arduino/issues/1981

rakyll, 2015-07-27:
I am working on a command line tool that wraps arduino, my primary goal is to give users no graphical output rather than disabling the graphics mode entirely -- my users are likely going to be using the tool on a desktop env anyways.

As it's mentioned on #1981, graphics dependencies are deeply baked in Base.java. But what about closing the splash screen right after the launch, before it's too late at https://github.com/arduino/Arduino/blob/2c05841588313918a3fcab7ecbc01023a8621f6f/app/src/processing/app/Base.java#L437?

carlosperate, 2015-07-27:
I believe that is currently only a problem on os x, this issue might have more info about it: https://github.com/arduino/Arduino/issues/1970

Maybe one of the main developer can shed some light on this specific request.

Since you are also working on a  CLI tool might be work checking if errors might trigger a window to pop up. I've found this issue on https://github.com/arduino/Arduino/issues/3515 , might be same in other operating systems.

matthijskooijman, 2015-07-29:
This is actually a duplicate of #1970, so I'm closing this report in favor of that one. IIRC, the problem is that the splash screen is shown by the loader which runs *before* any of our Arduino Java code runs.

closed 2015-07-29 by matthijskooijman