arduino/Arduino #3596: Signature of AT90USB82 and AVR_ATmega8U2 swapped in Descriptors.h

Reuti, 2015-07-28:
While having problems to use the DFU in the USB Serial Light Adapter with the correct target in dfu-programmer, I realized that the definition in the source code:

/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/firmwares/atmegaxxu2/arduino-usbdfu/Descriptors.h (in 1.0.6)

/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/firmwares/atmegaxxu2/arduino-usbdfu/Descriptors.h (in 1.6.5r3)

is wrong in lines 120-129 and the supplied compiled images for UNO-dfu_and_usbserial_combined.hex and MEGA-dfu_and_usbserial_combined.hex too:

		#elif defined(__AVR_AT90USB82__)
			#define PRODUCT_ID_CODE               0x2FEE
			#define AVR_SIGNATURE_1               0x1E
			#define AVR_SIGNATURE_2               0x93
			#define AVR_SIGNATURE_3               0x89
		#elif defined(__AVR_ATmega8U2__)
			#define PRODUCT_ID_CODE               0x2FF7
			#define AVR_SIGNATURE_1               0x1E
			#define AVR_SIGNATURE_2               0x93
			#define AVR_SIGNATURE_3               0x82

According to http://www.atmel.com/Images/doc7618.pdf page 2 and the datasheets http://www.atmel.com/images/doc7707.pdf page 246 resp. http://www.atmel.com/images/doc7799.pdf page 249 these definitions need to be swapped. Easiest by changing the defined() statement:

		#elif defined(__AVR_ATmega8U2__)
			#define PRODUCT_ID_CODE               0x2FEE
			#define AVR_SIGNATURE_1               0x1E
			#define AVR_SIGNATURE_2               0x93
			#define AVR_SIGNATURE_3               0x89
		#elif defined(__AVR_AT90USB82__)
			#define PRODUCT_ID_CODE               0x2FF7
			#define AVR_SIGNATURE_1               0x1E
			#define AVR_SIGNATURE_2               0x93
			#define AVR_SIGNATURE_3               0x82

facchinm, 2015-07-29:
Fixed in LUFA 140928; however, this issue is only related to older boards since the microcontroller is now a 16u2 (starting from UNO R3)

facchinm, 2015-08-10:
Closing and keeping track on #3597

Reuti, 2015-08-10:
https://github.com/abcminiuser/lufa/issues/58 i.e. in LUFA 140928 it's not completely fixed.

facchinm, 2015-08-10:
Sorry @Reuti , I completely missed the signature difference (I was blinded by the PRODUCT_ID_CODE swap :smile: ). However, I only closed the issue to make it easier to track lufa related problems, but since this one is not completely solved I'll close the other one and reopen this.

Reuti, 2015-08-10:
Thx, I though so. The 0x93 vs. 0x94 mismatch was newly introduced sometime between the LUFA 100807 used in Arduino and 140928.

closed 2018-10-03 by facchinm