arduino/Arduino #3597: Signature in DFU bootloader can't be delivered correctly by Arduino-usbdfu.c

Reuti, 2015-07-28:
File in question is /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/firmwares/atmegaxxu2/arduino-usbdfu/Arduino-usbdfu.c

According to http://www.atmel.com/Images/doc7618.pdf page 18, there are 4 bytes in total which resemble the complete signature:

30h Read Manufacturer Code
31h Read Family Code
60h Read Product Name
61h Read Product Revision

These can also requested by the dfu-programmer (via the get command as: manufacturer | family | product-name | product-revision). While the first two can be accessed correctly, Arduino-usbdfu.c fails to deliver the 3rd correctly while fails to deliver the forth at all, as it's not defined in Descriptors.h (in the same directory).

In Lines 726-727 in Arduino-usbdfu.c only 30h is subtracted in all cases, and output of some bytes ahead for the index of 60h and 61h happens:

	else if (IS_ONEBYTE_COMMAND(SentCommand.Data, 0x01))                    // Read signature byte
	  ResponseByte = SignatureInfo[DataIndexToRead - 0x30];

A case should be added to deliver at least the Product Name correctly.

facchinm, 2015-07-29:
Hi @Reuti , both this issue and #3596 has been resolved by LUFA in the meantime (arduino-dfu is based on LUFA's DFU bootloader, version 100807). For example this one becomes 
```
	else if (IS_ONEBYTE_COMMAND(SentCommand.Data, 0x01))                    // Read signature byte
	{
		if (DataIndexToRead < 0x60)
		  ResponseByte = SignatureInfo[DataIndexToRead - 0x30];
		else
		  ResponseByte = SignatureInfo[DataIndexToRead - 0x60 + 3];
	}
}
```
in LUFA 140928. We will fix both the issues, however from a normal user's point of view it's very difficult to flash the updated dfu bootloader so it will be mainly useful for newly produced boards. 

Thanks for reporting!

Reuti, 2015-07-30:
Hi,

well - I only had a look at the source I found in the Arduino download for now. But it’s good to know that it was already fixed. Question: shouldn't it be + 2 in the statement below, not + 3?

ResponseByte = SignatureInfo[DataIndexToRead - 0x60 + 3];

And I think the complete field definition in LUFA 140928:

const uint8_t SignatureInfo[4]  = {0x58, AVR_SIGNATURE_1, AVR_SIGNATURE_2, AVR_SIGNATURE_3};

should be rearranged to put 0x58 at the end. AVR_SIGNATURE_1 is always 0x1E for Atmel (not Family Code [as far as I know for now]). Where we might need a constant would be the Product Revision.

I'll raise these on the LUFA list and post the result here.

-- Reuti

facchinm, 2015-08-10:
Solved mainline, keeping track on #3596

closed 2015-08-10 by facchinm