arduino/Arduino #3598: Arduino preprocessing stripping comments when creating .cpp file

WestfW, 2015-07-29:
Why has the IDE started stripping comments when preprocessing .ino files to .cpp files in the build directory?  This is liable to make disassembly and real debugging more difficult (for those few who do it), and violates the principal of doing as little pre-processing as possible, and I don't see any advantage to doing it...

Chris--A, 2015-07-29:
Yup very bad, it doesn't actually do it properly:

```arduino

//This is a valid comment. \
so is this, but not today!

void setup() {}
void loop() {}
```

http://stackoverflow.com/questions/7059549/c-multi-line-comments-using-backslash

ffissore, 2015-08-05:
Sorry, this issue belongs to a long list of issue tagged with https://github.com/arduino/Arduino/labels/Component%3A%20Preprocessor

Chris--A, 2015-08-05:
Ah, I had forgotten I had a PR for this too: https://github.com/arduino/Arduino/pull/3240

And this is a duplicate of #956

ffissore, 2015-09-11:
Fixed by #3779

closed 2015-09-11 by ffissore