arduino/Arduino #3600: variable 'BuzzerMorseCodes_str' must be const in order to be put into read-only section by means of '__attribute__((progmem))'

zfaizal, 2015-07-30:
Arduino: 1.6.5 (Linux), Board: "Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"

Build options changed, rebuilding all
In file included from /home/z/Downloads/arduino-1.6.5/hardware/arduino/avr/cores/arduino/Arduino.h:28:0,
                 from arduinotx_buzz.h:17,
                 from arduinotx_buzz.cpp:14:
arduinotx_buzz.cpp:26: error: variable 'BuzzerMorseCodes_str' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
 PGM_P BuzzerMorseCodes_str[] PROGMEM = {
                              ^
variable 'BuzzerMorseCodes_str' must be const in order to be put into read-only section by means of '__attribute__((progmem))'

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

facchinm, 2015-07-30:
Hi @zfaizal , your problem is related with the "new" compiler (from IDE 1.6.0) not accepting PROGMEM as a standalone keyword. The right way to solve the issue is explained here https://github.com/arduino/Arduino/wiki/1.6-Frequently-Asked-Questions#errors-related-to-avr-progmem-changes

VaclavSal, 2015-07-30:
variable 'BuzzerMorseCodes_str' must be const in order to be put into read-only section by means of 'attribute((progmem))'

or you cannot use variable to store in PROGMEM.

Also check for correct  current syntax  - it changed.

zfaizal, 2015-07-30:
Hi @facchinm. I found the page you referred to earlier, and attempted to reinstall all libraries as well as add const after PGM_P, but it results in "invalid conversion from 'const char* const*' to 'const char**' [-fpermissive]"

zfaizal, 2015-07-30:
Hi @VaclavSal where do I find the correct syntax?

VaclavSal, 2015-07-30:
It has been a while - I stop using PROGMEM when I switched to Due.
If I remember I would have included it in my post, sorry .
somethink like  
const char PROGMEM 
or 
char const PROGMEM 
or PROGMEM char 

I think  when you change the variable to const the compiler MAY tell if the syntax is correct.
Sorry, but OF just don't remember stuff well.

Famous last words - Google it .

VaclavSal, 2015-07-30:
Found ONE! 

const char bmp_header[54] PROGMEM =

Chris--A, 2015-07-31:
The Arduino reference also has some examples: https://www.arduino.cc/en/Reference/PROGMEM

facchinm, 2015-08-10:
Closing as solved

closed 2015-08-10 by facchinm