arduino/Arduino #3601: IDE Tab organisation for large projects with lots of files

mypinballs, 2015-07-30:
Hi, I would like to request an improvement to how tabs are displayed once you get a large project and the tabs go past your screen size. Some extra settings for multiple rows of tabs would be great for large multi filed projects. I like splitting up my large projects into more manageable files, but the file count can get high (+20) pretty quickly, so you end up only being able to see half you files on the tab line.

Cheers 

Jim

PaulStoffregen, 2015-07-30:
Someday when the IDE supports breakpoint & stepping debug, a way to display (perhaps as read-only) the code from libraries and the core library files will be needed.  Horizontal tabs won't be up to the task.  Implementing a good way to view many tabs/files now will help pave the way for a debug-enabled future.

lmihalkovic, 2015-10-30:
No matter what, the popup menu always lets you switch to the source code, even if the tab is gone...

@PaulStoffregen  a couple days ago I made the [following changes](https://github.com/arduino/Arduino/issues/4050) to my local IDE: to avoid the disappearing tabs issue, I am dividing my project into libraries which are stored in the main sketch. The code for the main level as well as the libraries can be edited directly in the IDE. Switching around is also straightforward with the outline tree view. computers are fast these days so it would be easy to extend it with say, the name of the methods/classes in each source file and some basic data to jump to them directly (we are talking regex, not ast).

lmihalkovic, 2015-11-21:
@PaulStoffregen  I'm at the stage where I can edit local libraries and display the system one. The tabs are still somewhat the originals, but because I can split a project's code into sub-folders, then the right sliding does not matter as much... I am also looking into re-ordering them because I get frustrated when writing code if I can't have it the way I like. all that's missing from your list is a debugger. is this something you already started working on? is that something you'd be interested in collaborating on? it is getting high on my remainders list.

orange-cloud, 2015-11-21:
Is there any progress on adding what i original asked for? Better organisation of tabs surely would be a benefit for all users regardless. Having the ability to group tabs in to sub sets may be another way to handle it? IDE improvements are so needed right now!

lmihalkovic, 2015-11-21:
From what I have gathered, the company has very little bandwidth available for big items. Arduino Create seems to be receiving a lot of their focus, and seemingly small items for the IDE are starting to get closed with a short "Arduino Create will do that". I started a separate [master-list](https://github.com/lmihalkovic/Arduino/issues/1) referencing the issues the team is either not interested in or does not have the resources to look into. This is one of them.

orange-cloud, 2015-11-21:
I still like the idea of a desktop IDE with a debugger, decent tab handling, better library updating and continuing improvements suggested by the community and then put into new releases. I don't see the need to move form this type of model. Its working nicely

lmihalkovic, 2016-01-07:
although this is not a complete solution to this problem, something like the following may be a affordable interim step: 

[image]

cycle through the file names while the ALT key is pressed and only switch to the desired one when all keys are released.

again it represents a minimal effect as it can all be done in a single new class and with 2 lines changed inside Editor (will work even in @matthijskooijman's proposed cleaned-up Editor code)

lmihalkovic, 2016-01-07:
```
Someday when the IDE supports breakpoint & stepping debug, a way to display (perhaps
as read-only) the code from libraries and the core library files will be needed
```

In my experience so far I think the same scheme works fine when displaying the source code for any of library. 

[image]

I guess the real test would be to open one of these sketches with many source files and see how things work out.