arduino/Arduino #3602: IDE configuration lacks a dump command

Insiduous, 2015-07-30:
It would be extremely helpful in porting projects (and in using makefiles) to have the IDE either dump the configuration <source:key=value> information. Either with a commandline option, or as part of the /tmp file building of an ino/project.

The https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification page simply does not explain enough and the dump should be a trivial matter.

Insiduous, 2015-07-30:
For example:
{runtime.platform.path} appears to be IDE generated, (unclearly); 
{build.path} and {build.arch} are not said to be IDE generated, but must be (why not list them together?)

The list is too long and confusing to bother regexing out of the sources and data. Simply generate a (on stdout or to a project tmp/ file as you build a project) something like:

IDE:runtime.platform.path=<whatever>
:
boards.txt:uno.name=Arduino Uno

Please consider above.

matthijskooijman, 2015-07-31:
Try the `--get-pref` option, which should include `runtime.platform.path` for the currently selected board as well. See chttps://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc

It seems the build variables like `build.path` and `build.arch` are not included there, since those are only set during the build, I believe.

Does this help for your usecase?

Insiduous, 2015-07-31:
It's incomplete, and it doesn't show WHERE things came from, so - no.. (It also needs to be sorted, which did help readability). It's something I can pass on to others, but it won't do the job, (which is irritating because it's incomplete AND it really could be dumped in the tmp dir with the other build noise).

At this point, it's "better than nothing" - and that phrase is getting old.

matthijskooijman, 2015-07-31:
I think part of the problem is that the preferences system is getting a bit convoluted - there's a lot of sources of these preferences (preferences.txt, platform.txt, boards.txt, the IDE code, etc) and these values are both proper preferences and runtime variables. Properly documenting the entire thing might be a start, but perhaps the system needs an overhaul to make more sense (causing all kinds of backward compatility problems, of course...)

PaulStoffregen, 2015-07-31:
What does "very useful for porting projects as a package" mean?

Really, what good will some sort of export of internal stuff be?  What will you actually *do* with it?  Well, other than open more issues asking for more functionality to somehow import or use it?

Insiduous, 2015-07-31:
matthijskooijman: I suspect proper documentation would be a huge win, but there is an awful lot of information-hiding, obfuscation and obtuseness involved, (making things 'easier' for non-programmers and making programmers confused/lost).

PaulStoffregen: Actually, it never occured to me to ask it be imported back IN.. It's the application STATE, (and if we are saving it while 'verifying a sketch', the compile-state [and upcoming upload state] as well). In conjunction with the boards.txt, platforms.txt, and "3rd party hardware" documentation, it would be useful in converting an IDE-sketch to a Makefile package. Further, now that I think about it - it would be useful in assisting folks get their stuff running in assorted discussion/help channels and forums. Additionally, it would work hand in hand with the 3rd-party HW documentation.

PaulStoffregen, 2015-07-31:
Those are laudable goals.  But dumping lots of internal state to a text file is probably only a small piece of the total work needed to generate a good makefile or compose a useful problem report or generate readable documentation.

Are you planning to seriously work on any of these?

Insiduous, 2015-07-31:
I have zero interest in Java, so no. From the text-dump, helping others would be easier, and ascertaining the requirements of a Makefile for their projects would be easier. A "problem report" is overkill at this point, and the broken documentation has been broken for quite awhile.