arduino/Arduino #3603: The Arduino credits webpage is too hard to find, yet very useful--add it to the "What is Arduino" link options

ElectricRCAircraftGuy, 2015-07-30:
Website issue.
The credits webpage is too hard to find. It is a very useful page, as it recognizes past contributors, including Processing and Wiring, and it helps to explain what is Arduino. 

Also, there is much political debate about Arduino.cc not recognizing Wiring and Processing enough, whereas Arduino.org does (ex: http://www.arduino.org/about-us, one click away from their entrance page). 

### Simply adding the credits page as a 4th option at the top of arduino.cc, when you go to arduino.cc and click "What is Arduino", would be very helpful. It would make a lot of folks feel much better too.

VaclavSal, 2015-07-30:
IBFL

ElectricRCAircraftGuy, 2015-07-31:
No idea what that means. Google searches don't reveal much on this one.

mastrolinux, 2015-07-31:
It is one click away for us too. It is in the footer ;) It is better to demonstrate who we are with our work.

mastrolinux, 2015-07-31:
The link is in our footer. It is better to demonstrate who we are with our work.

q2dg, 2015-07-31:
@mastrolinux You should read issues with a little more care before closing them... @ElectricRCAircraftGuy is talking about THIS link: https://www.arduino.cc/en/Main/Credits , which is not in footer nor anywhere. I should add it there (or as @ElectricRCAircraftGuy says, as another section inside "What's is Arduino? page), and even write a link pointing to it in https://www.arduino.cc/en/Guide/Introduction

mastrolinux, 2015-07-31:
To me Credits it is the old version of https://www.arduino.cc/en/Main/AboutUs I could be wrong and will ask my colleagues but the goal looks pretty much the same.

q2dg, 2015-07-31:
Well, in "Credits" there a brief explanation about origins and also mentions Wiring, Processing, Ivrea, external designers of some Arduino boards (Sparkfun, Adafruit, Sharebot, Gravitech...)... I think it's different fron "About us", which shows current staff

ElectricRCAircraftGuy, 2015-07-31:
@q2dg, many thanks for posting the link and helping to get the issue reopened. In my attempt to write the concern concisely, I forgot to put the link at all!

So, let me return to my natural state of being more long-winded, and I'll provide a full argument to my case.

@mastrolinux , to me, the two pages (Credits: https://www.arduino.cc/en/Main/Credits), and (About us: https://www.arduino.cc/en/Main/AboutUs) are nothing alike. The former answers the question, "What is Arduino?", "Where did it come from?", "What is its history?", "Who/what influenced it to become what it is today?", etc. The latter simply answers the question, "Who is on the Arduino.cc team today?" These two pages are completely different. I think the Credits page actually answers the question "What is Arduino?" better than the 3 link windows that pop up currently when you click "What is Arduino" on the front page of arduino.cc. *Nevertheless, the 3 existing links are still useful, so I think adding the "Credits" page as a 4th link, and calling it "Background and History" or something to that effect, is imperative.* 

Let me further explain.  In the Arduino vs. Arduino drama today, on *nearly every* article I read, there are comments at the bottom of each article, by readers, stating something to this effect:

"Interesting article but again, as most of the other articles about Arduino, it misses a critical point by not mentioning the real pioneers that made Arduino possible:
– Processing was developed by Casey Reas and Ben Fry and provides the IDE all the boards are using.
– Wiring was developed as a thesis by Hernando Barragán and provided the framework used by all the boards.
Arduino.ORG pays tribute to both of them openly, while Arduino.CC stays mum."
You can find this comment here: http://hackaday.com/2015/07/23/hackaday-interviews-federico-musto-of-arduino-srl/, by Rei Vilo. You can notice that I (panther3001 there) responded.

I have clearly chosen sides (see here for example: http://www.amazon.com/review/R3PXVMNU7Z7049/ref=cm_cr_dp_title?ie=UTF8&ASIN=B00US73ZIO&channel=detail-glance&nodeID=541966&store=pc), and I first learned of Arduino from the Credits link we've posted above. However, the Credits link is gone now. I cannot find it *anywhere* on the Arduino site. The only place I can find it is by doing a google search for "arduino credits"--but that requires that I know beforehand what I'm searching for--or by opening up the Arduino IDE and reading the splash screen that comes up while it's opening, as it actually has a link to the credits page. 

This is a highly political environment. It makes us (Arduino.cc) look really bad when they (Arduino.org) describe the history of Arduino....and we hide it on a credits page that isn't linked to anywhere---while the credits page explains "What is Arduino" better than the "What is Arduino" link we currently have on Arduino.cc.

*In Summary:
The Credits page and the About us page are completely different. Having the former without the latter is ok, but would be sad not to mention the current team. Having the latter without the former would be *wrong*, as it doesn't mention the foundation or history of what made Arduino great. Havinb both would be...ideal.*

q2dg, 2015-07-31:
I totally agree

mastrolinux, 2015-08-01:
Hi,

Thanks for your support. This is why the link https://www.arduino.cc/en/Main/AboutUs contains a link to https://www.arduino.cc/en/Main/Credits in the main menu on the left with Red font to better show it. It was so since the AboutUs page was created.

q2dg, 2015-08-01:
Ups! Sorry!

ElectricRCAircraftGuy, 2015-08-05:
@mastrolinux, thanks for those links. I never found the Credits page that way...I suppose it didn't stick out. 

I still would like to see the Credits page added as a link to help answer "What is Arduino?" I see this issue has the label "Wontfix", however. Is that still the case, or do you plan on implementing my suggestion? 

Let me know and I'll close the issue if necessary.

JASON33, 2015-08-05:
Been tooling with adruino new one smd. (Asus X551mav.1"""2 .ousb.then has
1..3.0.ports. Today it did auto 10 Windows.from8.1bing..loaded.  .and have
not tryed.to upload.10.Windows.do they have 3.0.blue arduino.port .
opinion????

JASON33, 2015-08-05:
Sorry.I'm.trying.old school..lol.I bought 2(2.4)ghz..addicore I will upload
.thru.chrome.or any one familiar 10.0bing

mastrolinux, 2015-08-05:
@ElectricRCAircraftGuy We are going to review the header of our websites in the next few months so there could be some changes. I would close the issue for now. If our UX designers will agree than we can reopen it and ask for suggestions.

Thanks a lot.

closed 2015-08-05 by mastrolinux