arduino/Arduino #3604: Arduino IDE Doesn't Use proxy.pac

McNeight, 2015-07-31:
Our company uses a proxy.pac file for configuring web browsers to access the Internet, and other proxy-aware clients can be manually configured. With the recent change in the IDE to remove proxy settings and use the system default, the IDE can no longer connect to the Internet from behind my company firewall. It doesn't recognize the proxy.pac file, and I can't manually enter the proxy settings.

As a test, I changed the Internet Options on my Windows 7 machine from "Use automatic configuration script" to "Proxy server" with the correct settings, and the IDE was able to connect to the Internet with no problem. However, this is not a viable long term solution.

The IDE needs to either be fully compatible with the system proxy settings, or have the ability to set a proxy manually restored.

Thanks.

-Neil

ffissore, 2015-08-03:
AFAIU, you manually configure each and every app you use to access internet through your proxy/firewall. Having a system wide setting, thus keeping your from manually configure every app, is not an option. Is there a technical reason?

McNeight, 2015-08-03:
Correct. For my company, it is ease of administration. Any time a proxy setting needs to be changed, it is simply changed in the proxy.pac file rather than attempting to update the proxy settings for 150,000+ computers. Firefox uses this file directly, while Internet Explorer and Chrome use this file through the Internet Options settings. Other proxy-aware applications such as PuTTY or Git, I have to configure manually.

Within my company, I'm also fortunate to have administrator access to my computer. For other users, changing this setting may be impossible. Until now, I have had no reason to deviate from the default settings, but if I do I am on my own for any problems I encounter.

From what I've seen, many applications will check the system settings first, and then fall back to program settings if present. With the IDE, since there are no longer any program settings, it is either system settings affecting every program or nothing at all.

ffissore, 2015-08-05:
@McNeight is your proxy.pac a local file or do you specify an internal url from where downloading it?

McNeight, 2015-08-05:
It's an internal URL.

ffissore, 2015-08-07:
@McNeight please monitor PR #3637. @ArduinoBot will provide urls for downloading a version of the IDE with a (hopefully feature complete) proxy settings panel

McNeight, 2015-08-09:
I'm on it. Thanks!

ffissore, 2015-09-01:
Fixed with #3637

closed 2015-09-01 by ffissore