arduino/Arduino #3605: Update of libraries from library manager window idea

orange-cloud, 2015-07-31:
I was updating some of my own libraries recently and was thinking that it would be good if you could specify a location for an update of a chosen library. i.e search for your library in the library manager then have an update button that asks you to choose where there new library is, then copies the library over to the arduino libraries folder and updates the details.

jbhome, 2015-07-31:
Sounds like a great idea but maybe have the install library cleanup old and
install new rather than two. It's unlikely one would want two versions
installed

orange-cloud, 2015-07-31:
Yes, absolutely. The workflow should just replace the old library with the new one, or maybe have an 'old' folder where the old library is moved so the ide interface can rollback if you so wish?

VogonJeltz, 2015-07-31:
+1
IMO the first place to store libraries is the Arduino install folder, e.g.
c:\program files (x86)\Arduino\libraries
and of course, uninstall existing older libs before.

Also uninstalling either libs by the library manager  should be provided!

ffissore, 2015-08-04:
See Sketch > Include Libraries ... > Add .ZIP library. You can specify a zip file containing a library. IDE will unpack it into your SKETCHBOOK/libraries folder

closed 2015-08-04 by ffissore

orange-cloud, 2015-08-04:
So how does this work in terms of updating an existing library? What happens to the existing folder? Does the name of the folder need to be kept the same to avoid duplicates? Can an update button within the library manager's item list itself not be added?