arduino/Arduino #3607: Change SD Card

nemeier, 2015-08-01:
In some of my Projects I need to change the SD card. There is a simple allow this. I just require a single line in the SD.c library.
```
boolean SDClass::begin(uint8_t csPin) {
  /*
    Performs the initialisation required by the sdfatlib library.
    Return true if initialization succeeds, false otherwise.
   */
// ----> Line Added
  if (root.isOpen()) root.close();      // allows repeated calls
// <--- Line Added
  return card.init(SPI_HALF_SPEED, csPin) &&
         volume.init(card) &&
         root.openRoot(volume);
}
```
After changing the SD card all you need is to reinitialize the sd card.

Will properly solve ISSUE #1103

dzzie, 2015-12-27:
this fixed an issue i was having thanks!

bborncr, 2016-03-18:
This fix worked for me also. Is there any possible problem that could result from this patch?

agdl, 2016-07-12:
This issue was moved to arduino-libraries/SD#5

closed 2016-07-12 by agdl