arduino/Arduino #3611: Windows 10 not recognizing Arduino Micro or Leonardo as HID

cyborg5, 2015-08-02:
If you have an Arduino Micro or Leonardo running a sketch that does mouse and keyboard emulation, it will not work under Windows 10 after any reboot. You can unplug and replug the device and it will work but if it is left in during a reboot or an initial power on, it is not properly recognized.

Looking at the device manager after a reboot, Windows 10 detects it as a USB COM instead of for example Arduino Micro. Apparently that is why it does not see it as a Human Interface Device. If you upload a sketch with mouse and/or keyboard libraries the device manager then correctly identifies the device as Arduino Micro or Leonardo and the HID features work okay.

This problem is critical for many disabled people such as myself because we rely on specialized Arduino powered devices to assist us in accessing a PC. I don't know anything about USB drivers or HID USB protocols but apparently Windows 10 is sending some sort of different query than previous versions of Windows used and thus it is not fully identifying the device's capabilities.

NicoHood, 2015-08-16:
Windows10 uses its built in drivers if no other driver is installed. This could be a problem.
IDE 1.6.6 Added Pluggable HID which also causes problems if the device suddenly changes from Mouse to Keyboard. This is a known issue of Windows.

Maybe this helps?
https://github.com/NicoHood/HID/wiki/Troubleshoot-FAQ#switching-the-hid-core

cyborg5, 2015-08-16:
I read the troubleshooting link that you sent and really didn't understand most of it. Uninstalling and reinstalling the device doesn't help. Even if I tell Windows to remove the driver and then reinstall it, the problem persists after reinstalling. This has occurred on two different PCs that were working fine under Windows 8.1. One of them has a device that is an Arduino Micro and the other is an Arduino Leonardo. Upon every reboot of the PC the device is identified as a USB serial device. Reloading a sketch, unplugging and replugging the device, or pushing the devices reset button each can get Windows tend to properly identified as Micro or Leonardo. Somehow Windows 10 is querying the device differently than it did in 7 or 8.1 on the initial boot up.

 

https://github.com/NicoHood/HID/wiki/Troubleshoot-FAQ#switching-the-hid-core

NicoHood, 2015-08-16:
USB2 or 3 Host? With usb hub and what USB version (2/3) is the hub? (yep this is a serious problem/question).

Oh I think I know why. This might be a problem in the USB-Core. A problem that was always ignored.
https://github.com/arduino/Arduino/pull/3640#issuecomment-130786857

The correct way is this:
https://github.com/NicoHood/HID/blob/2.2/avr/cores/hid/USB-Core/USBCore.cpp#L99-L115
https://github.com/NicoHood/HID/blob/master/avr/cores/hid/USB-Core/USBCore.h#L117-L135

NicoHood, 2015-08-16:
Try this fix (change your arduino core file manually with the new)
https://github.com/arduino/Arduino/compare/master...NicoHood:patch-5

Program your arduino ALWAYS with an HID device then. If this works we need to patch it properly so it changes the ID flexible if CDC is used or not.

cyborg5, 2015-08-16:
You are assuming a level of sophistication on my part but unfortunately I do not reach :-) To answer your previous questions, both devices were plugged directly into the PC not through a hub. I believe that they are USB 3.0 because both machines are relatively new but I'm not sure if either my desktop or laptop have a combination of USB 2.0 and 3.0. I know some of my ports are 3.0 maybe all of them. I'm not sure where I would apply the patches that you have suggested. And if I understand your reply correctly it means that these devices would always have to be programmed with a sketch wherein they are HID devices. While 99% of the time my particular devices would only be in that capacity, I don't want to make any patches that would limit the usability of these devices for other purposes. I'm not the only person who has experienced this problem. There are other reports in the Arduino forums so this problem should be easily reproducible by someone who is more knowledgeable than I am about USB protocols.

 

 <https://github.com/arduino/Arduino/compare/master...NicoHood:patch-5> master...NicoHood:patch-5

NicoHood, 2015-08-16:
For testing. Its only for testing that you should use this patch, to ensure the patch works, because the patch is made for HID devices, not without. If it works we can make the patch better to work with all kind of settings. You want a fix, you need to test it since I do not have windows. I am also no Arduino member, I am just trying to help you with my knowledge, nothing more.

Go to arduino.cc and download the latest arduino nightly build as zip. Extract it.
Replace the file in `Arduino/hardware/arduino/avr/cores/arduino/USBCore.cpp` with this file:
https://raw.githubusercontent.com/NicoHood/Arduino/9cc7dadb86e92bab817c924c0c6e0a7c5fb6a53a/hardware/arduino/avr/cores/arduino/USBCore.cpp

Restart the IDE, upload an HID program and test if it works.

cyborg5, 2015-08-16:
I did not mean to leave the impression I was ungrateful for your help. I can however follow these instructions. I just didn't know what to patch or how. This I can do. I will back up previous file so that I can easily reverse it if necessary. I will report back soon.

 

https://raw.githubusercontent.com/NicoHood/Arduino/9cc7dadb86e92bab817c924c0c6e0a7c5fb6a53a/hardware/arduino/avr/cores/arduino/USBCore.cpp

NicoHood, 2015-08-17:
Since you need to download the nightly IDE there is no need for a backup. After all tests you can just delete the nightly IDE (use the non install zip), use the stable 1.6.5 again and you are fine.

(it would be nice if you, or someone else could remove this email overhead of your comments on github)

cyborg5, 2015-08-18:
Great news! I downloaded a daily build this afternoon and recompiled my keyboard and mouse sketches. I had to add 
```
#include <Mouse.h>
#include <Keyboard.h>
#include <HID.h>
```
in order to get them to compile. But other than that they compiled and now they work correctly. This was tested on my laptop using Arduino Micro and on my desktop using Arduino Leonardo. Both machines were originally Windows 8.1 recently upgraded to Windows 10 which was causing the problem. The problem is not resolved on both machines. Presumably once IDE 1.6.6 goes live this will be permanently fixed. For now the recompile is doing just fine.

Many thanks for all your help.

closed 2015-08-18 by cyborg5

NicoHood, 2015-08-18:
Without my fix???
That is weird. @facchinm have you changed anything that could have fixed that?

cyborg5, 2015-08-18:
I just double checked to make sure I was doing this right. The problem persists using IDE 1.6.5 but compiling with IDE 1.6.6 from 9:42 AM yesterday with no modifications except to add the includes I mentioned and it works just fine. Did not add any patches in addition to what was already in that daily build.

NicoHood, 2015-08-18:
Thats is weird but positive somehow. I still like to know what caused the problem.

NicoHood, 2015-08-24:
Can you try to include the Mouse AND the Keyboard at the same time? You dont need to use both, just include the .h file and see if the issue is there or not.

If the issue is still there, try to use my patch. I am still wondering what solved/caused this issue.

cyborg5, 2015-08-24:
Using IDE 1.6.5 I do not include Mouse.h nor Keyboard.h nor HID.h because they aren't necessary. In fact I don't think they exist I'm not sure. When compiling with IDE 1.6.6 my sketch would not compile because it did not have value such as KEY_ESC or MOUSE_LEFT. So I had to include them and then it would work okay. My particular sketch required both because it does both mouse and keyboard emulation.

NicoHood, 2015-08-24:
So you used mouse and keyboard at the same time and it worked? Sounds good but still no idea about the cause.

adicontakt, 2017-02-15:
I have the same problem but I can"d solved , everytime I have to unplug the usb , any solution? I try to compile ,to add
#include <Mouse.h>
#include <Keyboard.h>
#include <HID.h>
but I get a lot of errors

facchinm, 2017-02-15:
@adicontakt 
`#include <HID.h>` doesn't need to be added, but it should compile cleanly on boards with USB support (Micro, Leonardo, etc). Are you sure you selected the correct board from "Boards" menu?

adicontakt, 2017-02-15:
yep , becouse I can upload other project . 
this is the original code
`#include <Encoder.h> // The rotary encoder library http://www.pjrc.com/teensy/td_libs_Encoder.html
int accel = 1;
int c = 0 ;
// This is an acceleration factor. Use between 1 and 8 to suit how quickly the volume goes up or down.
Encoder myEnc(2, 3);
long oldPosition = 0;

void setup() {
; 
   //Serial.begin(9600);
}
void loop() {
  long newPosition = myEnc.read() / accel;
  if (newPosition != oldPosition) {
    if(newPosition > oldPosition) { 
    Remote.increase(); //  Увеличить громкость
    delay(50); 
    Remote.clear();   
    }
    if(newPosition < oldPosition) {  
    Remote.decrease(); 
    delay(50); 
    Remote.clear(); 
    }
    
    oldPosition = newPosition;
    //Serial.println(newPosition);
    
  }
  if (digitalRead(16) == LOW && c==0) { 
   
    Remote.mute(); 
     
    delay(10); 
    Remote.clear(); 
    c=1;  
    delay (100);
  }
  if (digitalRead(16) == HIGH && c==1) { 
    c=0;  
    delay (100);
  }
}`
an this is what i get after I compile (with the mention that the board work OK (except I have to reconnect the board after restart)
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring.c: In function 'init':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring.c:264:3: warning: #warning Timer 2 not finished (may not be present on this CPU) [-Wcpp]

  #warning Timer 2 not finished (may not be present on this CPU)

   ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring.c:273:3: warning: #warning Timer 2 not finished (may not be present on this CPU) [-Wcpp]

  #warning Timer 2 not finished (may not be present on this CPU)

   ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\Print.cpp: In member function 'size_t Print::print(const __FlashStringHelper*)':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\Print.cpp:44:23: warning: '__progmem__' attribute ignored [-Wattributes]

   const char PROGMEM *p = (const char PROGMEM *)ifsh;

                       ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\Tone.cpp:210:12: warning: #warning this may not be correct [-Wcpp]

[... 174 lines cut ...]

 class Keyboard_ : public Print

       ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:100:0,

                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:193,

                 from sketch\Encoder_volume_control.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/USBAPI.h:122:7: error: previous definition of 'class Keyboard_'

 class Keyboard_ : public Print

       ^

Multiple libraries were found for "Mouse.h"
 Used: C:\Users\adico\OneDrive\Documente\Arduino\libraries\Mouse
 Not used: C:\Program Files (x86)\Arduino\libraries\Mouse
Multiple libraries were found for "Keyboard.h"
 Used: C:\Users\adico\OneDrive\Documente\Arduino\libraries\Keyboard
 Not used: C:\Program Files (x86)\Arduino\libraries\Keyboard
exit status 1
Error compiling for board Arduino Leonardo.

facchinm, 2017-02-15:
It looks like you are following this guide https://forum.arduino.cc/index.php?topic=309025.msg2146241#msg2146241 or something similar. The core modification needed for `Remote` to work is https://github.com/stefanjones/Arduino/commit/bc12d7697206ea6ddaf5ebc708b8d7aade831251, which is based on older 1.0.5 core and looks unmaintained.
For something up-to-date try the great [HID Project](https://github.com/NicoHood/HID) from @NicoHood which exposes all the nice USB stuff in a pluggable way.

adicontakt, 2017-02-15:
but my board is promicro(leonardo) , not UNO

cyborg5, 2017-02-15:
It looks like the majority of your problems are coming from the fact that the IDE is finding two copies of the various libraries such as Mouse.h and Keyboard.h. One of them on your hard drive and one in your One Drive folders. Anytime you have 2 library files that are identical you're going to get those conflicts.

It seems that your problems are mostly related to configuring the IDE and don't have anything directly to do with IRLib or my example code. I suggest you go to one of the support forms at Arduino.cc and seek their advice on how to avoid these conflicts.

adicontakt, 2017-02-15:
maybe if I delete the one of the libryary?and keep only one
btw how to use  HID from nicohood? delete the hid that comes from arduino and add this library?
I don"t wanna go on arduino forum becouse i am a newbie and they don"t have ebough time to explane me how to do it.

facchinm, 2017-02-16:
@adicontakt the forum is exactly the place to help newbies, and it's a shame that you didn't find any help there.
My advice is to start from the wiki pages (https://github.com/NicoHood/HID/wiki/API-Documentation and https://github.com/arduino/Arduino/wiki/PluggableUSB-and-PluggableHID-howto), install the newest IDE, specify another sketchbook from the preferences panel so you have a clean setup without duplicate libraries etc. 

That said, this repo tracks bugs with cores and the IDE itself, so we try to keep it tidy and avoid too much user support. Writing on the forum may be frustrating sometimes but it's the right place for this kind of problems.

adicontakt, 2017-02-18:
so basically you say to delete everything I have now from arduino , install the new IDE , add HID from NicoHood and after that upload my sketch(who actually is not my sketch , is taken from web)

facchinm, 2017-02-20:
@adicontakt ,
if the sketch is old you'll need to slightly modify it but yes, it's the most safe way (so you son't end up with libraries duplicates and non-standard core modifications)

adicontakt, 2017-02-20:
ok , so I delete everything , i reinstaled 
the first error:Media was not declared in this scope
so i added the "encoder lib. from the website of author of the sketch 
no i get this error "
Arduino: 1.8.1 (Windows 10), Board: "Arduino Leonardo"

C:\Users\adico\Desktop\teste teste\Encoder_volume_control\Encoder_volume_control.ino: In function 'void setup()':

Encoder_volume_control:9: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
    Keyboard.begin(); 

    ^

C:\Users\adico\Desktop\teste teste\Encoder_volume_control\Encoder_volume_control.ino: In function 'void loop()':

Encoder_volume_control:16: error: 'Remote' was not declared in this scope

     Remote.increase(); //  ĐŁĐ˛ĐµĐ»Đ¸Ń‡Đ¸Ń‚ŃŚ ĐłŃ€ĐľĐĽĐşĐľŃ?Ń‚ŃŚ

     ^

Encoder_volume_control:21: error: 'Remote' was not declared in this scope

     Remote.decrease(); 

     ^

Encoder_volume_control:32: error: 'Remote' was not declared in this scope

     Remote.mute(); 

     ^

exit status 1
'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?

after that I include keyboard lib and get this error"
Arduino: 1.8.1 (Windows 10), Board: "Arduino Leonardo"

C:\Users\adico\Desktop\teste teste\Encoder_volume_control\Encoder_volume_control.ino: In function 'void loop()':

Encoder_volume_control:18: error: 'Remote' was not declared in this scope

     Remote.increase(); //  ĐŁĐ˛ĐµĐ»Đ¸Ń‡Đ¸Ń‚ŃŚ ĐłŃ€ĐľĐĽĐşĐľŃ?Ń‚ŃŚ

     ^

Encoder_volume_control:23: error: 'Remote' was not declared in this scope

     Remote.decrease(); 

     ^

Encoder_volume_control:34: error: 'Remote' was not declared in this scope

     Remote.mute(); 

     ^

exit status 1
'Remote' was not declared in this scope

and i am stuck ? who is this "remote" ? is I use stefan method to modify the files in core it works but with the proble that i have to reinsert the usb cable everytime on restart