arduino/Arduino #3612: can't add library by manager, no internet connection (???), cant find USB class to include

VogonJeltz, 2015-08-02:
I can't add  a library by the manager, 
says no internet connection (??? rubbish!!! ), 
cant find USB class to include, nothing to find on https://www.arduino.cc/en/Reference/USBHost

ffissore, 2015-08-04:
It seems this is going to be solved by manually updating the lib. See #3622.

closed 2015-08-04 by ffissore

VogonJeltz, 2015-08-04:
HELL, CAN'T YOU READ ???
I CAN'T UPDATE MANUALLY BECAAUSE THE LIBS CAN'T BE FOUND ANYWHERE!!!!!

WHERE THE FU** ARE  Scheduler.h and USBhost.h to be found to be installed manually ?????

PLEASE LEARN TO READ AND TO UNDERSTAND BEFORE YOU'RE CLOSING ISSUES !!!

JASON33, 2015-08-04:
Same on mine says com is busy.raspberry. pi. New. Ima gonna loaded
perfect.ardiuno quality is down hilllllllll😞