arduino/Arduino #3613: IDE 1.6.1 + DUE: USBhost does not work with USB keyboard, no keyboard key press transmitted

VogonJeltz, 2015-08-02:
I jsu found USBhost lib in IDE 1.6.1, 
now compiling 1.6.3 with this installed lib  + DUE:  

no USB keyboard inputs are shown - why don't I get keyboard keys on Serial when I press a keyboard key?

(I just plugged a PC USB keyboard to the 2nd USB of the DUE )

Test sketch:

~~~
#include <KeyboardController.h>

// Initialize USB Controller
USBHost usb;

// Attach Keyboard controller to USB
KeyboardController keyboard(usb);

void setup(){
  Serial.begin(9600);
}

void loop(){
  usb.Task();
}

void keyPressed() {
  Serial.print("Pressed:  ");
  Serial.print(keyboard.getKey());
  Serial.println();
}

source reference: https://www.arduino.cc/en/Reference/KeyboardControllerConstructor
~~~

VogonJeltz, 2015-08-02:
ps,
also this alternative code shows NOTHING on Serial, except "program started":

~~~
/*
 Keyboard Controller HID Example

 Shows the output of a USB Keyboard connected to the USB
 controller of an Arduino Due Board.

 created 8 Oct 2012
 by Cristian Maglie
 */

// Require keyboard control library
#include <KeyboardController.h>

// Initialize USB Controller
USBHost usb;

// Attach keyboard controller to USB
KeyboardController keyboard(usb);

// This function intercepts key press
void keyPressed() {
  Serial.print("Pressed:  ");
  printKey();
}

// This function intercepts key release
void keyReleased() {
  Serial.print("Released: ");
  printKey();
}

void printKey() {
  // getOemKey() returns the OEM-code associated with the key
  Serial.print(" key:");
  Serial.print(keyboard.getOemKey());

  // getModifiers() returns a bits field with the modifiers-keys
  int mod = keyboard.getModifiers();
  Serial.print(" mod:");
  Serial.print(mod);

  Serial.print(" => ");

  if (mod & LeftCtrl)
    Serial.print("L-Ctrl ");
  if (mod & LeftShift)
    Serial.print("L-Shift ");
  if (mod & Alt)
    Serial.print("Alt ");
  if (mod & LeftCmd)
    Serial.print("L-Cmd ");
  if (mod & RightCtrl)
    Serial.print("R-Ctrl ");
  if (mod & RightShift)
    Serial.print("R-Shift ");
  if (mod & AltGr)
    Serial.print("AltGr ");
  if (mod & RightCmd)
    Serial.print("R-Cmd ");

  // getKey() returns the ASCII translation of OEM key
  // combined with modifiers.
  Serial.write(keyboard.getKey());
  Serial.println();
}

void setup()
{
  Serial.begin(115200);
  Serial.println("Program started");
  delay(200);
}

void loop()
{
  // Process USB tasks
  usb.Task();
}

what's going on here ?????????
~~~

VaclavSal, 2015-08-02:
If I recall correctly it did  work only on mouse , for me , which does not help you.
The KEY is Task() which is does not  return anything to loop(). 
 
You need to dig into "Task"  and retrieve "USB status", it is there somewhere. 
Task has two major if's() and those are relatively  easy to trace. 
I am currently not working on USB, but may get back to it soon.
Sorry I could not be more help.
Good luck.

VogonJeltz, 2015-08-02:
hmmmmh... 
[quote]The KEY is Task() which is does not return anything to loop(). [/quote]
this actually really sounds like a very very clever programmed source :-/
Amazing, that no one but be run into issues or at least caused rapid bug fixes so far.... :-/

As I have really no clue of C++ ...
what do you mean, where should I start to search...?

~~~
/* Copyright (C) 2011 Circuits At Home, LTD. All rights reserved.

This software may be distributed and modified under the terms of the GNU
General Public License version 2 (GPL2) as published by the Free Software
Foundation and appearing in the file GPL2.TXT included in the packaging of
this file. Please note that GPL2 Section 2[b] requires that all works based
on this software must also be made publicly available under the terms of
the GPL2 ("Copyleft").

Contact information
-------------------

Circuits At Home, LTD
Web      :  http://www.circuitsathome.com
e-mail   :  support@circuitsathome.com
*/

/* USB functions */

#include "Arduino.h"
#include "Usb.h"
#include <stdio.h>

static uint32_t usb_error = 0;
static uint32_t usb_task_state = USB_DETACHED_SUBSTATE_INITIALIZE;

/**
 * \brief USBHost class constructor.
 */
USBHost::USBHost() : bmHubPre(0)
{
	// Set up state machine
	usb_task_state = USB_DETACHED_SUBSTATE_INITIALIZE;

	// Init host stack
	init();
}

/**
 * \brief Initialize USBHost class.
 */
void USBHost::init()
{
	devConfigIndex	= 0;
	bmHubPre		= 0;
}

/**
 * \brief Get USBHost state.
 *
 * \return USB enumeration status (see USBHost::task).
 */
uint32_t USBHost::getUsbTaskState(void)
{
    return (usb_task_state);
}

/**
 * \brief Set USB state.
 *
 * \param state New USBHost status to be set.
[... 770 lines cut ...]
			if (rcode)
			{
				TRACE_USBHOST(printf("/!\\ USBHost::Task : USB_STATE_CONFIGURING failed with code: %lu\r\n", rcode);)
				if (rcode != USB_DEV_CONFIG_ERROR_DEVICE_INIT_INCOMPLETE)
				{
					usb_error = rcode;
					usb_task_state = USB_STATE_ERROR;
				}
			}
			else
			{
				usb_task_state = USB_STATE_RUNNING;
				TRACE_USBHOST(printf(" + USB_STATE_RUNNING\r\n");)
			}
            break;

        case USB_STATE_RUNNING:
            break;

        case USB_STATE_ERROR:
            break;
    }
}

~~~

VaclavSal, 2015-08-04:
"As I have really no clue of C++ ."
Than be ready to receive many advices in line "USB is very complicated..." 
I am really not sure how I can help you ii it (keyboard) does not work for you and same time you feel it  (USBHost) is a good program. 
And the reasons nobody  else said anything about USB (keyboard) not working is because there is no interest in Arduino circles.

VogonJeltz, 2015-08-04:
I have no idea if USB host is a goog program as I haven't tried it yet.
 I simply want to read the keyboard just like I'm able to do it when a keyboard is plugged to my PC or to a RaspPi, actually that's all about it.

VogonJeltz, 2015-08-04:
at
https://github.com/arduino-libraries
a USBhst version is thankfully provided which works and returns keystrokes!
there are some issues though  
(RIGHT-SHIFT + characters => not uppercase, 
Backslash not printed, 
OR operator | not printed, but ~ instead )

VogonJeltz, 2015-08-06:
there are different other issues about the current USBhost lib mentioned in this thread:
https://github.com/arduino/Arduino/issues/3630

VogonJeltz, 2015-09-02:
When can we expect an improvement in the Keyboard functions, like e.g.
kbhit()
getch()
getche()
getchar()
gets()
scanf()

?

VogonJeltz, 2015-09-19:
bump!
when will there be an update for reasonable keyboard stdin C functionality?

per1234, 2017-07-03:
The remaining unsolved portion of this issue is a duplicate of https://github.com/arduino/Arduino/issues/3630

closed 2017-07-03 by per1234