arduino/Arduino #3615: 1.6.5 USB "not in sync" issue on OS X 10.10.3

WesseLab, 2015-08-03:
On 1.6.5 of the Mac version IDE, one cannot download a sketch to the target more than once without having to disconnect the USB cable and plug it back in.  The The error message is below:

Sketch uses 17,316 bytes (56%) of program storage space. Maximum is 30,720 bytes.
Global variables use 975 bytes (47%) of dynamic memory, leaving 1,073 bytes for local variables. Maximum is 2,048 bytes.
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x8e
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xc2
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x4f
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xe2
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x82
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x4b
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xec
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xc9
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x6e
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xe2

ffissore, 2015-08-04:
Please post your question on the [arduino forum](http://forum.arduino.cc/). This repository issues are for bugs and feature proposals for the Arduino IDE, not for support enquiries. Thank you for your understanding

closed 2015-08-04 by ffissore

WesseLab, 2015-08-04:
Sorry Federico.  I thought it was IDE releated, but it turned out to be
drivers.  My apologies for not closing it sooner. :(

Eric