arduino/Arduino #3616: Error in layout drawing for "Button" project

warry-williams, 2015-08-03:
arduino-1.6.5-r2-windows\arduino-1.6.5-r2\examples\02.Digital\Button
This path links to some documentation for the beginner's project "Button".
The "layout picture shows the input as pin 2. The schematic and sketch language assign this to pin 12. Not a big deal, but it could throw a newbie for a loop.

agdl, 2015-09-01:
@warry-williams i think you refer to these images https://www.arduino.cc/en/Tutorial/Button
In the page both the images and the sketch are consistent and use pin 2 so can you please link the wrong place?

warry-williams, 2015-09-01:
Thank you for taking the time to help. I have been in electronics many years but never used bread boards before- I expect I had something plugged incorrectly.

agdl, 2015-09-02:
@warry-williams it all is ok in the documentation, can you please close the issue?

agdl, 2015-09-03:
@warry-williams I'm closing this. Reopen it if needed please.

closed 2015-09-03 by agdl