arduino/Arduino #3617: Arduino Uno brown-out level too low?

matthijskooijman, 2015-08-03:
Someone on IRC just pointed out that the brown-out level of the Arduino Uno, at least the one specified by the [0x05 extended fuse in the boards.txt]() (not sure about manufactured boards) is 2.7V. Given that a 328 needs 3.6V to guarantee working at 16Mhz, this 2.7V sounds way too low. I double-checked the datasheet - an extended fuse of 0x04 would be better, which is a bod level of 4.3V (one step up).

matthijskooijman, 2015-08-03:
Note that I didn't check the fuse settings of any other board.

NicoHood, 2015-08-16:
16u2 Fuse is set to 3V.
32u4 Fuse is set to 2.4V
Your Link above is broken.

I guess some usb ports can easily drop below 4.3V which is still > 3.6V but < 4.3V (I normally have 4.4-4.5V) so this could be a problem?

matthijskooijman, 2015-08-16:
I fixed the link.

If the voltage drops below 4.3V but above 3.6V, this would mean that the Atmega would reset, while not strictly required. However, the current situation is that if the voltage drops below 3.6V (but not below 2.7V), the Atmega doesn't cleanly reset but could lock up, not recovering properly when the voltage raises to proper levels again.

I'd want to have extra clean resets rather than potential complete lockups, especially as a default.

NicoHood, 2015-08-16:
I think I had those situations already. Not sure if it was caused by the brown out detection but yeah we should change this. Then the fuses for 32u4 and 16u2 also needs to be changed.