arduino/Arduino #3619: UTFT_SdRaw

ghlawrence2000, 2015-08-03:
Submitted for inclusion in library manager.

https://github.com/ghlawrence2000/UTFT_SdRaw/tree/V1.2.0

ffissore, 2015-08-04:
Repo added but lib didn't go into the list because you did not make a release. Please double read the FAQ https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

ffissore, 2015-08-04:
Once properly tagged, the lib will be available within the hour

closed 2015-08-04 by ffissore

ghlawrence2000, 2015-10-30:
Hello, I made a new realease V1.2.2 on git of the same library and 15 hours later it has still not appeared in library manager, is there anything else I need to do?
The new git release is here :- https://github.com/ghlawrence2000/UTFT_SdRaw/releases/tag/v1.2.2

Thanks,

Graham