arduino/Arduino #3621: how to update libraries offline (e.g., downloading a complete lib zip file) ?

VogonJeltz, 2015-08-04:
as online lib updating by the lib manager does not work on my cPCs:
how to update libraries offline  by  downloading a complete lib zip file and then c+p the desired libs manually to the libraries system folder ?

matthijskooijman, 2015-08-04:
This works just like it worked before the library manager was present. See https://www.arduino.cc/en/guide/libraries

closed 2015-08-04 by matthijskooijman

VogonJeltz, 2015-08-04:
where can one find all the desired libs?
e.g., Scheduler, USBhost,...

VogonJeltz, 2015-08-04:
please don't close an issue if it's not 100% resolved!

matthijskooijman, 2015-08-04:
You can typically click "more info" in the library manager to get to the site / github page of the library. I think there is also some JSON file on arduino.cc somewhere, that contains all the info and download urls of the libraries shown in the manager, but I don't have the url handy. I suspect it is hardcoded in the java source somewhere. IIRC the url was also mentioned in the mailing list post that introduced the library manager a few months ago, perhaps you could have a look in the mailing list archives.

> please don't close an issue if it's not 100% resolved to me!

An issue that is closed can still be responded to and even re-opened if needed, so please just let us use the issue tracker in the way we think it is useful.

VogonJeltz, 2015-08-04:
no, please read more carefully before you close an issue !!!
The lib mnager don't work because it says "no internet connection" what is actually bullshit  !