arduino/Arduino #3626: where can libs like Scheduler and USBhost be found to be installed manually ?

VogonJeltz, 2015-08-04:
where can libs like Scheduler and USBhost be found to be installed manually ?
My lib manager don't work because it tells "no internet connection" what is actually bullshit, so where can I downlaod these libs manually in order to install it manually?
There expect 1 download link to exist  providing an overview and single downloads for all available libs somewhere!

Arjan-Woltjer, 2015-08-04:
The Usb host libraries are maintained by other people than the Arduino folks. You could have just searched "usb host shield" on github and would have found your library. After that you follow the instructions handed to you beforehand on this forum, e.g. look in the arduino documentation on how to install libraries manually.

VogonJeltz, 2015-08-04:
I actually doubt that, USBhost and Scheduler both were part of the 1.6.1 standard installation and now are missing in the 1.6.3

Arjan-Woltjer, 2015-08-04:
So hard to find..
https://github.com/felis/USB_Host_Shield_2.0

VogonJeltz, 2015-08-04:
this is NOT the USBhost lib I'm looking for, and it's not the Scheduler lib I'm looking for!

Both are expected to be found easily on a Arduino Download page for re-installation and completion of the standard installation which regrettably is longer complete since 1.6.1 !

Arjan-Woltjer, 2015-08-04:
Stop ranting and please make some effort to search yourself. Is this then what you are looking for? https://github.com/arduino-libraries/USBHost. Took me some 2 minutes, after    which you could have mentioned the defect to the arduino team.

VogonJeltz, 2015-08-04:
hell, don't you understand?
ok, ich schreibe es in deutsch, ich kann es offenbar in Englisch nicht klar machen:
der library manager funktioniert nicht,
es fehlen mir sämtliche libs, die nur vom library manager installiert werden können,
und sie sollen sich alle an 1 gemeinsamen Platz befinden!
nicht nur USBhost,
sonder auch Scheduler 
und auch alle  zehntausend restlichen libs, die nicht in der Standardinstallation enthalten sind, inklusive aller Baords und dem ganzen Rest!

Arjan-Woltjer, 2015-08-04:
Ich verstehe ein bisschen Deutsch. I get the point you are making, but I just tested the library manager, and am able to install all your mentioned libraries. Using W10, Arduino 1.6.5R2. On the other hand, above mentioned github location has the USBHost library and I'm sure the other libraries can be found there as well. Just install them in the libraries directory in the skechdirectory you assigned.

VogonJeltz, 2015-08-04:
bei mir geht es eben nicht!!
der library manager behauptet,  es gäbe keine Internet Verbindung, was absoluter Blödsinn ist, aber darurch kann ich jetzt überhaupt nichts mehr nachinstallieren!
Daher erwarte ich sämtliche libs an einem gesonderten Platz, übersichtlich auszuwählen, zum separaten Download!
Natürlich weiß ich wie man libs installiert, aber ich muss sie an 1 zenttalen Platz gemeinsam finden können!
Sie alle einzeln mühsam in dem github-Durcheinander wie Stecknadeln in einem Heuhaufen suchen zu müssen ist ein unzumutbarer Humbug!

Arjan-Woltjer, 2015-08-04:
You could go back in time just about a year, and the situation you're in now would be the way to go. Please do respect the effort put into the library manager. Also keep in mind that organising the user space e.g. libraries written by all kinds of people is a huge undertaking and things don't always work the way you want. B.t.w. did you try to use the newer 1.6.5r3 release?

VogonJeltz, 2015-08-04:
ich habe keine Lust, über Sinn oder Unsinn des Download Managers zu diskutieren, ich erwarte eine Möglichkeit zum manuellen Download und zur manuellen Nachinstallation OHNE diesen Download Manager!

Arjan-Woltjer, 2015-08-04:
Und ich habe keine lust dich weiter zu helfen.

per1234, 2015-08-04:
https://github.com/arduino-libraries/USBHost
https://github.com/arduino-libraries/Scheduler

VogonJeltz, 2015-08-04:
thank you, THAT was really helpful!
never heard of this download page
https://github.com/arduino-libraries
this is exactly I had been searching like a pin in a haystack!
thanks again!

closed 2015-08-05 by ffissore