arduino/Arduino #3629: A bit of refactoring of the WiFi code

mladenb, 2015-08-05:
Hi,

At this page: https://www.arduino.cc/en/Guide/ArduinoWiFiShield
there is a code example in the section named "Scan for available networks", which could be improved by a simple refactoring.
If we move the line "delay(10000);" to the bottom of the loop() method, we could get rid of the last three lines in the setup() method (because they are redundant now).

Hope this helps :)

Cheers!

ffissore, 2015-08-05:
/cc @cmaglie

cmaglie, 2015-08-20:
I've fixed the example in the repository.

@agdl it should be automatically fixed in the tutorial page? I still see the old example.

agdl, 2015-09-01:
@cmaglie the page was linked to a wrong place. It is ok now. Every mod will be automatically tracked.

commit 5b3d16fa (2015-08-20): Simplified ScanNetwork example

Fixes #3629
closed 2015-09-01 by agdl