arduino/Arduino #3630: USBhost: additional functions for keyboard appreciated

VogonJeltz, 2015-08-05:
for USBhost it would be fine to have additional functions to rea the USB keyboard:
kbhit()
getch()
getche()
getchar()
gets()

VogonJeltz, 2015-08-05:
FYI, I already tried this following, but it does not work:
~~~
/*
 USBkeyboard Controller Example
 // modified by HaWe 05-July-2015

 Shows the output of a USB USBkeyboard connected to
 the Native USB port on an Arduino Due Board.

 created 8 Oct 2012
 by Cristian Maglie

 http://www.arduino.cc/en/Tutorial/USBkeyboardController

 This sample code is part of the public domain.
 */

// Require USBkeyboard control library
#include <Scheduler.h>;
#include <KeyboardController.h>
USBHost usb;                         // Initialize USB Controller
KeyboardController keyboard(usb);    // Attach USBkeyboard controller to USB

#define ENTER  40

//=====================================================================================

int  keyPressed() {
  int key =  keyboard.getKey();
  if(key) return 1;
  else return 0;
}   

int  keyReleased() { 
  int key =  keyboard.getKey();
  if(key) return 1;
  else return 0;
}

//=====================================================================================

int getche_() {
  static int16_t oemkey=0, modkey=0, currkey=0, lastkey=0; 
  
  
  oemkey=keyboard.getOemKey();
  modkey=keyboard.getModifiers();
  if(oemkey==88) oemkey=40; // ENTER, ENTERnum
  
  //if(lastkey==40) strcpy(USBstring,"");
  // ESC   = 41
  // Enter = 40, 88
  // Space = 44
  // Tab   = 43
  // TabBack   = mod 2 + 43
  // Backspace = 42 
  // F1    = 58
  // F5    = 62
  // F9    = 66
  // F12   = 69
  // INS   = 73
  // HOME  = 74
  // PgUp  = 75
  // DEL   = 76
  // END   = 77
  // PgDn  = 78
  // RIGHT = 79
  // LEFT  = 80
  // DN    = 81
  // UP    = 82  

  if(oemkey==ENTER)     currkey='\n';      // ENTER
  else  if(oemkey==42)  currkey='\8';      // BACKSPACE
  else  if(oemkey==43)  currkey='\t';      // TAB
  else  if( oemkey==41) currkey='27';      // ESC key
  else  if((modkey==0)&&(oemkey== 50)) currkey='\\';   // BACKSLASH
  else  if((modkey==0)&&(oemkey==100)) currkey='\\';
  else  if((modkey==2)&&(oemkey==50))  currkey='|';    // PIPELINE
  else  if((modkey==2)&&(oemkey==100)) currkey='|';
  else  if((modkey==32)&&(oemkey==50)) currkey='|';
  else  if((modkey==32)&&(oemkey==100)) currkey='|';  
  
  else  if((oemkey>=58)&&(oemkey<=69)) {;} // Function key F1-F12
  else  if((oemkey>=73)&&(oemkey<=82)) {;} // CURS key INS-PgDn, RIGHT-UP  
  
  else  currkey=keyboard.getKey();                     // normal letter

  if(currkey>27) Serial.write(currkey);
  while(!keyReleased());
  
  return currkey;
  
  
}

//=====================================================================================

char * gets_(char * USBstr) {
  static  int16_t  currkey=0, lastkey=0; 
  static  char     chbuf[3];
  char  * strptr = USBstr;
   
  currkey = getchar();
  
  if(currkey==8) {                         // BACKSPACE
      if(strlen(USBstr)>0)  USBstr[strlen(USBstr)-1]='\0';
      else strcpy(USBstr,"");
  }
  else { sprintf(chbuf, "%c", currkey); strcat(USBstr, chbuf); }  
  
  return strptr; 
}   

//=====================================================================================

void setup()
{
  Serial.begin(115200);
  Serial.println("Program started");
  Scheduler.startLoop(USBloop);
  delay(200);
}

//=====================================================================================

void loop() {
   char USBkey; 

   if(keyPressed()) USBkey=getche_(); 
   yield();
   
}   

//=====================================================================================

void USBloop()
{
  // Process USB tasks
  usb.Task();  // strangely seems to call keyPressed() in background
  delay(5);
  
}

//=====================================================================================
//=====================================================================================
~~~

VogonJeltz, 2015-08-05:
currently I see no way for use inside a custom function:
- how to detect if the user has pressed a key on the keyboard or not
- only in case a key has been hit, how to assign it to a variable, if not, just skip and proceed with the next command

VogonJeltz, 2015-08-05:
the easiest way of course would be to have USB as stdin and then just read from stream by ANSI C stdio.h functions!

VogonJeltz, 2015-08-17:
When can we expect an improvement in the Keyboard functions, like e.g.
kbhit()
getch()
getche()
getchar()
gets()
scanf()

?

VogonJeltz, 2015-09-19:
**bump!**
when will there be an update for reasonable keyboard stdin C functionality?

bbx10, 2016-01-06:
@VogonJ @cmaglie 

>usb.Task has to be called repeatedly but then blocks program execution intermediately and repeatedly for many seconds

I submitted an issue and PR in the USBHost repo to remove the 5 second usb.Task() delay. This fix makes it much easier to add USB keyboard/mouse support to an existing project.

https://github.com/arduino-libraries/USBHost/issues/2

There is another issue/PR to add control key support for the keyboard.