arduino/Arduino #3633: some font cant be changed

34626, 2015-08-06:
Hello,

Im a visually impaired user and my eyes prefer white text and black background. I have now tried to moddify the theme.txt file in lib on Linux Ubuntu but it would be very nice if there was a theme that is dark with white text and black background..

ffissore, 2015-08-07:
We don't support theme switching at the moment. Did you succeed by modifying theme.txt?

34626, 2015-08-07:
Thanks for reply!

We tried to modify all text related into #FFFFFF and all background to
#000000 but the code you write in arduino is not included so thats still
unreadable (comments and some other stuff is included)

I send you a image of Arduino and the theme.txt as we have modifited it.

another thing is the buttons if there could be light and dark to choose..

On the image you see an example code called Blink and here we cant see the
delay values or the port number..

I would also recommend the menu buttons to be text or optimized for being
in dark theme.. (I was told what the buttons did otherwise i didnt know
what they would do..
i also added a image named flush where you can see a torrent client, but
where the buttons/tabs has their names next to them.. and my theme (Dark
Numix) is used to make the app darkish..
Best regards /
Med venlig hilsen
Kenneth Rasmussen

ffissore, 2015-08-07:
Indeed, you also need to tweak entries listed under `TEXT - LITERALS` and `TEXT - COMMENTS`

34626, 2015-08-07:
Just saw i forgot to send the theme.txt file.

-- 
Best regards /
Med venlig hilsen
Kenneth Rasmussen

#FUNCTIONS COLOR 			#D35400 - ORANGE			KEYWORD1
#FUNCTIONS COLOR			#D35400 - ORANGE			KEYWORD2
#STRUCTURE COLORS 			#5E6D03 - GREEN			KEYWORD3
#VARIABLES COLOR 				#00979C - BLUE				LITERAL1

#ERROR COLOR					#A61717 - RED
#COMMENTS // COLOR			#95A5A6 - LIGHT GREY
#COMMENTS /**/ COLOR		#434F54	- DARK GREY

# GUI - STATUS
status.notice.fgcolor = #FFFFFF
status.notice.bgcolor = #000000
status.error.fgcolor = #FFFFFF
status.error.bgcolor = #000000
status.edit.fgcolor = #FFFFFF
status.edit.bgcolor = #000000
status.font = SansSerif,plain,12

# GUI - TABS
# settings for the tabs at the top
# (tab images are stored in the lib/theme folder)
header.bgcolor = #17A1A5
header.text.selected.color = #005B5B
header.text.unselected.color = #007e82
header.text.font = SansSerif,plain,12

# GUI - CONSOLE
console.font = Monospaced,plain,11
console.font.macosx = Monaco,plain,10
console.color = #000000
console.output.color = #FFFFFF
console.error.color = #FFFFFF

# GUI - BUTTONS
buttons.bgcolor = #000000
buttons.status.font = SansSerif,plain,12
buttons.status.color = #FFFFFF

# GUI - LINESTATUS   
linestatus.color = #ffffff
linestatus.bgcolor = #000000

# EDITOR - DETAILS

# foreground and background colors
editor.fgcolor = #ffffff
editor.bgcolor = #000000

# highlight for the current line
editor.linehighlight.color=#e2e2e2
# highlight for the current line
editor.linehighlight=false

# caret blinking and caret color
editor.caret.color = #FFFFFF

# color to be used for background when 'external editor' enabled
editor.external.bgcolor = #000000

# selection color
editor.selection.color = #2d2d2d

# area that's not in use by the text (replaced with tildes)
editor.invalid.style = #FFFFFF,bold

# little pooties at the end of lines that show where they finish
editor.eolmarkers = true
editor.eolmarkers.color = #FFFFFF

# bracket/brace highlighting
editor.brackethighlight = false
editor.brackethighlight.color = #FFFFFF

# TEXT - KEYWORDS

# FUNCTIONS
editor.keyword1.style = #FFFFFF,bold
editor.data_type.style = #FFFFFF,bold

# METHODS
editor.keyword2.style = #FFFFFF,plain
editor.function.style = #FFFFFF,plain

# STRUCTURES
editor.keyword3.style = #FFFFFF,plain
editor.reserved_word.style = #FFFFFF,plain

# TEXT - LITERALS

# constants & datatypes
editor.literal1.style = #FFFFFF,plain
editor.literal_char.style = #FFFFFF,bold
editor.literal_string_double_quote.style = #FFFFFF,bold

# p5 built in variables: e.g. mouseX, width, pixels
editor.literal2.style = #FFFFFF,plain
editor.variable.style = #FFFFFF,plain
editor.reserved_word_2.style = #FFFFFF,plain

# http://www.arduino.cc/
editor.url.style = #FFFFFF,underlined

# e.g. + - = /
editor.operator.style = #FFFFFF,plain

# ?? maybe this is for words followed by a colon
# like in case statements or goto
editor.label.style = #FFFFFF,bold

# TEXT - COMMENTS
editor.comment1.style = #FFFFFF,plain
editor.comment2.style = #FFFFFF,plain

# LINE STATUS - editor line number status bar at the bottom of the screen
linestatus.font	   = SansSerif,plain,10
linestatus.height  = 20

ffissore, 2015-09-08:
Any update on this? Have you found your way through theme.txt options?

34626, 2015-09-08:
The needed options we was unable to find..

Best regards /
Med venlig hilsen
Kenneth Rasmussen

ffissore, 2015-09-08:
Have you modified stuff like `editor.literal2.style`, `editor.comment1.style` and so forth, as suggested in https://github.com/arduino/Arduino/issues/3633#issuecomment-128626172 ?

34626, 2015-09-08:
> and so forth, as suggested in #3633 (comment)

all lines that is in theme.php we have tried to modify..
Do you have succes making Arduino all black and all text white?