arduino/Arduino #3634: Ethernet.maintain() documentation is unclear

matthijskooijman, 2015-08-06:
When using the Ethernet library with DHCP, `Ethernet.maintain()` should regularly be called to renew the loop when needed. However, the current documentation doesn't entirely make clear when or how often this should happen (the current docs could be implying that calling `maintain()` always tries to refresh the DHCP lease, but it first checks if the current lease time is still valid). Also, the `Ethernet.begin()` docs do not mention this at all.

I would propose to add this to the [`Ethernet.maintain()` page](https://www.arduino.cc/en/Reference/EthernetMaintain) as a second paragraph under "description":

> You can call this function as often as you want, it will only re-request a DHCP lease when needed (returning 0 in all other cases). The easiest way is to just call it on every `loop()` invocation, but less often is also fine. Not calling this function (or calling it significantly less then once per second) will prevent the lease to be renewed when the DHCP protocol requires this, continuing to use the expired lease instead (which will not directly break connectivity, but if the DHCP server leases the same address to someone else, things will likely break).

Additionally, I would propose adding this to the last paragraph of the description of [`Ethernet.begin()`](https://www.arduino.cc/en/Reference/EthernetBegin):

> To make sure the DHCP lease is properly renewed when needed, be sure to call [`Ethernet.maintain()`](https://www.arduino.cc/en/Reference/EthernetMaintain) regularly.

Perhaps a second example would be helpful here?

The examples included with the library also don't use `Ethernet.maintain()`. From a quick grep at least `Webserver`, `UdpNtpClient` and `DhcpChatServer` would need to have a call added to their `loop()`. For bonus points, `DhcpAddressPrinter` should be modified to call `Ethernet.maintain()` and print a message and the new lease's details when the lease is renewed.

agdl, 2015-09-01:
@matthijskooijman I added the statements as you suggested. Let me know if they seem good to you!

matthijskooijman, 2015-09-01:
@agdl Documentation looks good to me. Did you also update the examples I mentioned?

agdl, 2015-09-02:
@matthijskooijman not yet I think I will do it this morning. I will make a PR and will link it to you ASAP

agdl, 2015-09-02:
@matthijskooijman here the mods https://github.com/agdl/Arduino/commit/65d7bd6ebba05174d1f4b96959ca82b072b3f51a

If they are ok for you I will make a PR. Let me know!

agdl, 2015-09-02:
@matthijskooijman see this 
https://github.com/agdl/Arduino/commit/be2c77f5000623788bb7af03c7899aadeee02b95

matthijskooijman, 2015-09-03:
@agdl Looks good to me. Two minor remarks:
 - Both commits should probably be squashed together for the PR, though the whitespace changes in the comment would be best kept in a separate commit.
 - There is some duplicate code to print the lease (once in setup, twice in loop). Putting that in a function might be nicer.

agdl, 2015-09-03:
@matthijskooijman done https://github.com/agdl/Arduino/commit/3f28c2452aaf6b61d2bceac5f6ea5458a585479f

If it is ok close this issue please!

closed 2015-09-03 by matthijskooijman