arduino/Arduino #3636: [DOC] Clarify PWM's max resolutions in Zero

q2dg, 2015-08-07:
As Arduino Zero has a 10-bit DAC, I suppose that's mean it will support only 10-bit PWM but on http://www.arduino.cc/en/Guide/ArduinoZero is said that "it can support up to 12-bit ADC and PWM resolutions..." So there is something which doesn't fall into place.

Moreover, I think https://www.arduino.cc/en/Reference/AnalogReadResolution and https://www.arduino.cc/en/Reference/AnalogWriteResolution needs a little make-up because they were written only with Due in mind.

Thanks

agdl, 2015-09-03:
@q2dg I think that the misunderstanding comes from the fact that we use the same API analogWrite() both for a DAC (really analog!) and a PWM operation (whose average can be considered analog). Which can be the best way to expose this in your opinion?

q2dg, 2015-09-03:
Well..simply indicating: "it can support up to 12-bit ADC, **12/10-bit???** PWM and 10-bit DAC resolutions..."

agdl, 2015-09-04:
@q2dg take a look now at:

https://www.arduino.cc/en/Guide.ArduinoZero
https://www.arduino.cc/en/Reference.AnalogReadResolution

please

q2dg, 2015-09-04:
Well, maybe in https://www.arduino.cc/en/Reference/AnalogWriteResolution should be added a paragraph like this (after the one related to Due):

The Zero has the following hardare capabilities:
    1x? pins which default to 8-bit PWM, like the AVR-based boards. These can be changed to 12-bit resolution.
    1 pin with 10-bit DAC (Digital-to-Analog Converter). By setting the write resolution to 10, you can use analogWrite() with values between 0 and 1023 to exploit the full DAC resolution

agdl, 2015-09-07:
@q2dg added as you suggested to do

q2dg, 2015-09-07:
Ok! Thanks a lot!

closed 2015-09-07 by q2dg