arduino/Arduino #3638: Rebuilding the IDE unter Ubuntu fails

NicoHood, 2015-08-07:
Ant run fails at the end with:
```
linux64-run:
     [exec] /usr/bin/env: bash
     [exec] : No such file or directory

BUILD FAILED
/home/linuxuser/Documents/Arduino/Arduino/build/build.xml:89: The following error occurred while executing this line:
/home/linuxuser/Documents/Arduino/Arduino/build/build.xml:604: exec returned: 127
```
I've installed and selected java8 with this tutorial:
http://ubuntuhandbook.org/index.php/2015/01/install-openjdk-8-ubuntu-14-04-12-04-lts/

A file named arduino is in /build/linux/work but I cannot start it via a shell too.

Used this tutorial:
https://github.com/arduino/Arduino/wiki/Building-Arduino

matthijskooijman, 2015-08-07:
Looks like you don't have "env" or "bash". What does `which env bash` output?

NicoHood, 2015-08-07:
```
/usr/bin/env
/bin/bash
```

matthijskooijman, 2015-08-07:
Something looks weird in your environment. I just tried on m y Debian system, where ant run works. In your output, `/usr/bin/env: bash: No such file or directory` looks like an error message that env cannot find bash. I'm not sure why it's spread over two lines, though. If I mess up my `build/linux/arduino` script, I get an error on a single line:

```
linux64-run:
     [exec] /usr/bin/env: foo: No such file or directory
```

Can you run `/usr/bin/env bash` manually? What is in `$PATH` in the terminal where you run `ant run`? Can you paste the first few lines of `build/linux/work/arduino`?

NicoHood, 2015-08-08:
Maybe elementary OS changed something. Its based on Ubuntu (based on Debian) but they might have changed something here, I am not sure.

/usr/bin/env bash in a terminal results in nothing. I guess this means it works, because I am in a shell again.

```
Arduino/build$ $PATH
bash: /home/user/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games: No such file or directory

---

Arduino/build$ head ./linux/work/arduino
#!/usr/bin/env bash

APPDIR="$(dirname -- "$(readlink -f -- "${0}")" )"

for LIB in \
    $APPDIR/java/lib/rt.jar \
    $APPDIR/java/lib/tools.jar \
    $APPDIR/lib/*.jar \
    ;
do
```

matthijskooijman, 2015-08-08:
@NicoHood aplied this patch (I talked to him through IRC) to print the environment values:
```
--- a/build/build.xml
+++ b/build/build.xml
@@ -603,7 +603,7 @@
   </target>
 
   <target name="linux64-run" depends="build" description="Run Linux (64-bit) version">
-    <exec executable="./linux/work/arduino" spawn="false" failonerror="true"/>
+    <exec executable="/usr/bin/env" spawn="false" failonerror="true"/>
   </target>
 
   <!-- Set '${dist_file}_available' property if toolchain dist_file is downloaded              -->
```

It shows that $PATH is correct:

    PATH=/home/linuxuser/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games 

So I'm puzzled about what's wrong here...

ffissore, 2015-08-10:
Once built (`ant clean build`) can you start IDE with `linux/work/arduino` ? Try tinkering with that bash script directly instead of making it run by `ant`

NicoHood, 2015-08-10:
Build Successfull but:

```
Arduino/build/linux/work$ ls
arduino          dist      hardware    lib        reference      tools
arduino.desktop  examples  install.sh  libraries  revisions.txt
Arduino/build/linux/work$ ./arduino 
: No such file or directory
```

matthijskooijman, 2015-08-10:
@NicoHood, I'm wondering if there is something weird with the `arduino` file. Could you post the output of `hd arduino | head` ? Also, if you create a script your self, using either `#!/usr/bin/env bash` or `/bin/bash` directly as the shebang, doest that run correctly?

NicoHood, 2015-08-11:
```
00000000  23 21 2f 75 73 72 2f 62  69 6e 2f 65 6e 76 20 62  |#!/usr/bin/env b|
00000010  61 73 68 0d 0a 0d 0a 41  50 50 44 49 52 3d 22 24  |ash....APPDIR="$|
00000020  28 64 69 72 6e 61 6d 65  20 2d 2d 20 22 24 28 72  |(dirname -- "$(r|
00000030  65 61 64 6c 69 6e 6b 20  2d 66 20 2d 2d 20 22 24  |eadlink -f -- "$|
00000040  7b 30 7d 22 29 22 20 29  22 0d 0a 0d 0a 66 6f 72  |{0}")" )"....for|
00000050  20 4c 49 42 20 69 6e 20  5c 0d 0a 20 20 20 20 24  | LIB in \..    $|
00000060  41 50 50 44 49 52 2f 6a  61 76 61 2f 6c 69 62 2f  |APPDIR/java/lib/|
00000070  72 74 2e 6a 61 72 20 5c  0d 0a 20 20 20 20 24 41  |rt.jar \..    $A|
00000080  50 50 44 49 52 2f 6a 61  76 61 2f 6c 69 62 2f 74  |PPDIR/java/lib/t|
00000090  6f 6f 6c 73 2e 6a 61 72  20 5c 0d 0a 20 20 20 20  |ools.jar \..    |
```

I have scripts like this which work:
```
#!/bin/bash
./arduino 2>&1 | tee arduinolog.txt
```

Now I replaced it with the 2nd thing you mentioned and I get
```
/usr/bin/env: /home/user/bin/arduinolog: Text file busy
```

NicoHood, 2015-08-16:
Any ideas? the env program doesnt seem to work.

matthijskooijman, 2015-08-16:
The problem seems to be that env actually does seem to be working, though not inside the build system (though that "text file busy" error is weird, not sure what that means or what text file (== linux jargon for a program, I think) is busy exactly).

Could you try changing the hasbang to `#!/bin/bash` in build/linux/work/arduino and see if that does run?

NicoHood, 2015-08-16:
In the test above my script was inside ~/bin, not inside the build path.

I created a file `test.sh` inside `/build/linux/work` with the content:
```bash
#!/bin/bash
echo test
```

The I ran:
```bash
/Arduino/Arduino/build/linux/work$ chmod 755 ./test.sh
/Arduino/Arduino/build/linux/work$ ./test.sh 
test
```

I changed the file to:
```bash
#!/usr/bin/env
echo test
```
And the error appears again.

I edited `linix/work/arduino` like this:
```bash
#!/bin/bash

APPDIR="$(dirname -- "$(readlink -f -- "${0}")" )"
...
```

Result:
```bash
/Arduino/build/linux/work$ ./arduino
bash: ./arduino: /bin/bash^M: bad interpreter: No such file or directory
/Arduino/build/linux/work$ sh ./arduino
: not found2: ./arduino: 
: not found4: ./arduino: 
./arduino: 6: ./arduino: Syntax error: word unexpected (expecting "do")
/Arduino/build/linux/work$ chmod 755 ./arduino
/Arduino/build/linux/work$ ./arduino
bash: ./arduino: /bin/bash^M: bad interpreter: No such file or directory
```

matthijskooijman, 2015-08-16:
Ah, it seems that your `arduino` script uses \r\n newlines, also confirmed by the previous hexdump you showed (that would also make some sense of the extra newline in the original error message). In my version of the script, there is just \n (0a) line endings:

$ hd build/linux/work/arduino|head -n 2
00000000  23 21 2f 75 73 72 2f 62  69 6e 2f 65 6e 76 20 62  |#!/usr/bin/env b|
00000010  61 73 68 0a 0a 41 50 50  44 49 52 3d 22 24 28 64  |ash..APPDIR="$(d|

Also, the original file shipped in the git repo has just \n:

$ hd build/linux/dist/arduino|head -n 2
00000000  23 21 2f 75 73 72 2f 62  69 6e 2f 65 6e 76 20 62  |#!/usr/bin/env b|
00000010  61 73 68 0a 0a 41 50 50  44 49 52 3d 22 24 28 64  |ash..APPDIR="$(d|

How does that file look for you? Does git show it as modified? Do you perhaps have git configured to do some line ending translation?

NicoHood, 2015-08-16:
I have git configured for windows lines, yes. Because all my commits before were for windows and then git showed the whole file was replaced. So I kept windows line endings. And git automatically changes this I think.

Can I change this setting for a single branch only? Not sure if this affects my PRs for the /hardware folder.

NicoHood, 2015-08-16:
It works!
I changed the line setting to "input" redownloaded the ide via git clone and it now works!!!

Someone should note this in the wiki page. Should I do this?

matthijskooijman, 2015-08-16:
> I have git configured for windows lines, yes. Because all my commits before were for windows and then git showed the whole file was replaced. So I kept windows line endings. And git automatically changes this I think.

I think this just means that your editor is configured to (always) write Windows line endings. Having it just preserve the original line ending style would be the best way to prevent massive changes to files, but I'm not sure if your editor supports that.

> Someone should note this in the wiki page. Should I do this?

Yeah, feel free!

I think we can close this issue?

NicoHood, 2015-08-16:
Its not the editor, its git which causes this issue. I'll edit it.

Thanks for the help!

closed 2015-08-16 by NicoHood

matthijskooijman, 2015-08-16:
@NicoHood But it is your editor that caused you to change the git line ending configuration.

NicoHood, 2015-08-16:
I did not touch any of the core files with an editor. I manually set the lineending to true some month ago. It was my fault to set it to true instead of input.

matthijskooijman, 2015-08-16:
You said:

> I have git configured for windows lines, yes. Because all my commits before were for windows and then git showed the whole file was replaced.

I assumed this means that your editor always writes Windows line endings, so when you change something in a file, all lines are marked as changed (which is solved by letting git translate from/to Windows line endings automatically). I might have misinterpreted what you said, though.