arduino/Arduino #3645: Wrong Avrdude config for atmega8u2

NicoHood, 2015-08-08:
The atmega8u2 avrdude config setting is wrong.
The bug was discovered here:
https://github.com/NicoHood/HoodLoader2/issues/6#issuecomment-128993103

Look at the screenshot and you will see some values are swapped. the correct values are:

page_size = 128
num_pages = 64
blocksize = 128

We could upload firmwares to a 8u2 with at90usb82 settings and -F switch. But with normal 8u2 setting we get a verification error. the head of the 8u2 block has also wrong values in the commented section where it describes the changes.

This file is affected (version 6.1)
http://svn.savannah.nongnu.org/viewvc/trunk/avrdude/avrdude.conf.in?root=avrdude&view=log

I cannot find the file in the IDE but could you please patch it?
It'd also be nice if someone could open an bug at the avrdude website. I dont have an account and dont want to create one, but this bug should be fixed, and it can be fixed fast.

Or other question:
Is it possible to use the default arduino core for a variants board and input a custom avrdude.conf file (to fix the issue for now permanently in my board definitions)

facchinm, 2015-08-10:
Hi Nico, thanks for reporting. The repo for our `avrdude` patches is [toolchain-avr] (https://github.com/arduino/toolchain-avr).
I think we'll move to a newer version quite soon, so we'd better report the issue to avrdude devs and solve upstream.

facchinm, 2015-08-10:
Reported upstream https://savannah.nongnu.org/bugs/index.php?45727

NicoHood, 2015-08-13:
Thanks for reporting it upstream!

Is there a way to pass a different avrdude config to the IDE?
I dont want to add big workarounds to my project just because of that.

Do you know when we can expect an ardude update for upstream?

NicoHood, 2015-08-16:
May anyone add this here and try it out?
https://github.com/arduino/toolchain-avr/tree/master/avrdude-patches

```
--- avrdude.conf.in	2013-09-18 07:59:06.000000000 +0200
+++ avrdude.conf.in_new	2015-08-16 10:54:52.764718970 +0200
@@ -12232,8 +12232,7 @@
 # Changes against ATmega16U2 (beside IDs)
 #    memory "flash"
 #        size            = 8192;
-#        page_size       = 64;
-#        blocksize       = 64;
+#        num_pages       = 64;
 
 part
     id               = "m8u2";
@@ -12320,8 +12319,8 @@
     memory "flash"
         paged           = yes;
         size            = 8192;
-        page_size       = 64;
-        num_pages       = 128;
+        page_size       = 128;
+        num_pages       = 64;
         min_write_delay = 4500;
         max_write_delay = 4500;
         readback_p1     = 0x00;
@@ -12353,7 +12352,7 @@
 
 	mode		= 0x41;
 	delay		= 6;
-	blocksize	= 64;
+	blocksize	= 128;
 	readsize	= 256;
       ;
 ```

I used avrdude 6.0.1 (not 6.1) and made the changes. Not sure if the other patches have to be applied before this patch, I started from the original avrdude file.

nerdralph, 2015-08-16:
@NicoHood Joerg seems to be the only person maintaining avrdude:
http://www.sax.de/~joerg/
You could email him your patch and see if it gets his attention.

NicoHood, 2015-08-16:
Thx, Done.
Dont we need to upgrade then to avrdude 6.1+ anyways? The IDE currently relies on 6.0.1

per1234, 2017-07-03:
These values are now correct in the current version of AVRDUDE included with the Arduino IDE. Thanks!

closed 2017-07-03 by per1234

NicoHood, 2017-07-03:
@per1234 Do you have any link to the patch? I've emailed him long time ago but I never got a notice that this got fixed. Good to hear it though.

per1234, 2017-07-03:
I don't see this fix in the patches. I guess maybe they are only published for some fixes?

However, you can see the bug report (https://savannah.nongnu.org/bugs/index.php?45727) was marked fixed Sat 31 Oct 2015 and the official version of AVRDUDE 6.3 from http://download.savannah.gnu.org/releases/avrdude/ has the changes you requested made to its avrdude.conf. I didn't check AVRDUDE 6.2 so I don't know if the fix made it into that release.

Note that Joerg Wunsch commented:
>Actually, the bug came in from the datasheet, while Atmel's
XML files show the correct values. Thus, you also ought to
open a bug report with Atmel so they might fix the datasheet.

I didn't look at the datasheet to see if it's been corrected yet.

NicoHood, 2017-07-03:
I wrote atmel multiple times to correct their datasheet for several bugs. Its like /dev/null so good to see an avrdude fix here.