arduino/Arduino #3646: I18N & L10N for Library and Board Managers.

Myriads, 2015-08-09:
Main IDE is well I18Nized and L10Nized. 
But Library Manager and Board Manager are some lacked of I18N and L10N: title, menu items and  descriptions are not supported with local languages.

ffissore, 2015-08-10:
Translations have been recently updated. Try the hourly build http://www.arduino.cc/en/Main/Software#hourly. Koren is almost complete https://www.transifex.com/mbanzi/arduino-ide-15/ but an additional review is always welcome

Myriads, 2015-08-10:
[image]
[image]

I attached 2 pictures of Library Manager. 
Type and subject are almost English. And more bad things are that all items are described in English.

ffissore, 2015-08-10:
Ah I see. Thanks

ffissore, 2015-11-18:
Those labels are now translatable. Please join our community of translator at https://www.transifex.com/mbanzi/arduino-ide-15/ide-15/

commit b450a274 (2015-11-18): Allowing boards/libs types and categories to be translated. Fixes #3646
closed 2015-11-18 by ffissore