arduino/Arduino #3654: Serial monitor does not honor some characters

xxxajk, 2015-08-10:
I'm not sure if these have been requested to be acted upon properly in the past or not, but I'm pretty sure somebody should have complained at some point in the past.

Notably, \r and \b are ignored.
```\r``` should simply return to the beginning of the line.
```\b``` should do a backspace, without a rubout.
These two are fairly important to make progress displays. I'm not asking for full cursor controls, but these two would be very helpful to have and shouldn't be a huge leap to implement, considering ```\t``` does work.

ffissore, 2015-08-10:
IDE Serial monitor is not aware of control chars other than \n. If you wish to use them, you need to use a terminal app like putty

closed 2015-08-10 by ffissore

xxxajk, 2015-08-10:
I use cu or screen, but that's not really the point.

VogonJeltz, 2015-08-28:
I double this TOP!
It's not a good practise by you, ffissore , to always ignore wishes about useful improvements in a stubborn manner and always forthwith setting them to "Wontfix"! 
Please note desires of the Arduino users with the proper respect, benevolence, and openness!

ffissore, 2015-08-28:
I do. Don't hesitate to provide a pull request

VogonJeltz, 2015-08-28:
I don't know and don't understand what a pull request is - Google translate does not help ;)

PaulStoffregen, 2015-08-28:
A pull request means you do all the coding work to implement the feature, then after you have carefully tested your code, you send your change using git's "pull request" feature.  Obviously you need to learn at least the basics of using the git software, as well as doing the actual work on the source code!

https://help.github.com/articles/using-pull-requests/

VogonJeltz, 2015-08-28:
sorry, I'm just a stupid low-end user with just 1 year Sketch coding experience in some spare time.
If I knew how to do it I already had long since made without losing a word about it.
But I daresay I know what stupid low-end users need for coding Sketches in some spare time.

PaulStoffregen, 2015-08-28:
Well, at least you now know what "provide a pull request" means!

VogonJeltz, 2015-08-28:
yes, and hopefully someone will pull this thing in future!

xxxajk, 2015-08-28:
At some point I may do this. I am currently busy in a contract job. It is
good to know that it is wanted.

yes, and hopefully someone will pull this thing in future!

PaulStoffregen, 2015-08-28:
If you do, please test CPU usage carefully with a Teensy 3.1 and Arduino Due sending continuous data without delay.  About a year ago, I put a tremendous amount of work into making Arduino not lock up and not run out of RAM under those conditions.  Let's not revert back to the bad-old-days of issue #2233 just to parse terminal control characters!

xxxajk, 2015-08-28:
Of course I would.

VogonJeltz, 2015-08-29:
xxxajk, remember, actually PaulStoffregen is number one motivator for innovations to the Arduino IDE.

xxxajk, 2015-08-29:
I know Paul well enough to know this ;-)

per1234, 2017-10-04:
Partial duplicate (\r) of https://github.com/arduino/Arduino/issues/346.