arduino/Arduino #3655: Arduino zero ethernet lib not compiling

Letreguilly, 2015-08-10:
just try the example WebServer sketch : 

/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp: In member function 'int EthernetClass::begin(uint8_t*)':
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:19:45: error: no matching function for call to 'SPIClass::beginTransaction(int, SPISettings)'
   SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
                                             ^
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:19:45: note: candidate is:
In file included from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/utility/w5100.h:13:0,
                 from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:1:
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note: void SPIClass::beginTransaction(SPISettings)
   void beginTransaction(SPISettings settings);
        ^
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note:   candidate expects 1 argument, 2 provided
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:30:47: error: no matching function for call to 'SPIClass::beginTransaction(int, SPISettings)'
     SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
                                               ^
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:30:47: note: candidate is:
In file included from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/utility/w5100.h:13:0,
                 from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:1:
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note: void SPIClass::beginTransaction(SPISettings)
   void beginTransaction(SPISettings settings);
        ^
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note:   candidate expects 1 argument, 2 provided
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp: In member function 'void EthernetClass::begin(uint8_t*, IPAddress, IPAddress, IPAddress, IPAddress)':
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:68:45: error: no matching function for call to 'SPIClass::beginTransaction(int, SPISettings)'
   SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
                                             ^
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:68:45: note: candidate is:
In file included from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/utility/w5100.h:13:0,
                 from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:1:
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note: void SPIClass::beginTransaction(SPISettings)
   void beginTransaction(SPISettings settings);
        ^
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note:   candidate expects 1 argument, 2 provided
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp: In member function 'int EthernetClass::maintain()':
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:89:51: error: no matching function for call to 'SPIClass::beginTransaction(int, SPISettings)'
         SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
                                                   ^
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:89:51: note: candidate is:
In file included from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/utility/w5100.h:13:0,
                 from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:1:
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note: void SPIClass::beginTransaction(SPISettings)
   void beginTransaction(SPISettings settings);
        ^
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note:   candidate expects 1 argument, 2 provided
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp: In member function 'IPAddress EthernetClass::localIP()':
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:107:45: error: no matching function for call to 'SPIClass::beginTransaction(int, SPISettings)'
   SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
                                             ^
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:107:45: note: candidate is:
In file included from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/utility/w5100.h:13:0,
                 from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:1:
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note: void SPIClass::beginTransaction(SPISettings)
   void beginTransaction(SPISettings settings);
        ^
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note:   candidate expects 1 argument, 2 provided
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp: In member function 'IPAddress EthernetClass::subnetMask()':
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:116:45: error: no matching function for call to 'SPIClass::beginTransaction(int, SPISettings)'
   SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
                                             ^
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:116:45: note: candidate is:
In file included from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/utility/w5100.h:13:0,
                 from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:1:
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note: void SPIClass::beginTransaction(SPISettings)
   void beginTransaction(SPISettings settings);
        ^
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note:   candidate expects 1 argument, 2 provided
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp: In member function 'IPAddress EthernetClass::gatewayIP()':
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:125:45: error: no matching function for call to 'SPIClass::beginTransaction(int, SPISettings)'
   SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
                                             ^
/home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:125:45: note: candidate is:
In file included from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/utility/w5100.h:13:0,
                 from /home/letreguilly/Documents/arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:1:
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note: void SPIClass::beginTransaction(SPISettings)
   void beginTransaction(SPISettings settings);
        ^
/home/letreguilly/.arduino15/packages/arduino/hardware/samd/1.6.0/libraries/SPI/SPI.h:101:8: note:   candidate expects 1 argument, 2 provided
Erreur lors de la compilation.

aethaniel, 2015-08-11:
shouldn't this issue moved to https://github.com/arduino/ArduinoCore-samd/issues ?

agdl, 2015-09-23:
yes @aethaniel you are right. I'm closing this,. Please @Letreguilly post this as indicated by @aethaniel.

closed 2015-09-23 by agdl