arduino/Arduino #3658: Port access taking twice as long on Mega 2560 with 1.6x builds

drp0, 2015-08-11:
I have written a program to utilise the 1.2MHz ADc TLV571.
This is a parallel ADC.
With build 1-5.6-R2, I can read the TLV571 at a top rate of 1.142 Mhz
This is consistent with the stated speed of the ADC chip.
With  a 1.6x build the maximum rate is 549.5 Khz.

I use direct port access to initiate the capture, then poll a parallel port line for a completion. 
The max rate is achieved with variable delayer set to zero.
variable econ = 8
I have tried:
 without disabling the interrupts:
 without the delay.
This suggests that the issue is with the V1.6 method used to directly access the port.

void usedelay(){
cli();
  while(triggered==false){
  PORTC = B00000010; // RD low, WR high, CS low
  PORTC = B00000110; // RD high, WR high, CS low
    while ((PINC & econ) == 0 ); // wait for econ to go high
  newval=PINA;
  trip = newval-oldval;
    if(!trigplus) trip = -trip;
    if (trip > trigger) triggered=true; else oldval=newval;
  }
starttime=micros();
  for (i=0;i<BUF_SIZE;i++){
  PORTC = B00000010; // RD low, WR high, CS low
  cli();
  PORTC = B00000110; // RD high, WR high, CS low
  bufa[i]=PINA;
  sei();
  delayMicroseconds(delayer);
  }
endtime=micros();
}

ffissore, 2015-08-11:
(edited for better code formatting)

drp0, 2015-09-14:
After much experimentation!
The issue lies with the global declaration of the unsigned int i.
If this variable is replaced with a local unsigned int, void nodelay() works perfectly, with data capture exactly as expected:
This begs the question, what has changed between build 1.5 and 1.6, in the implementation of global variables?
On a positive note, the local variable processing time improves the sample rate by 43KHz.

void nodelay(){
unsigned int il;
cli();
  while(triggered==false){
  PORTC = B00000010; // RD low, WR high, CS low
  PORTC = B00000110; // RD high, WR high, CS low
    while ((PINC & econ) == 0 ); // wait for econ to go high
  newval=PINA;
  trip = newval-oldval;
    if(!trigplus) trip = -trip;
    if (trip > trigger) triggered=true; else oldval=newval;
  }
starttime=micros();
  for (il = 0; il < BUF_SIZE; il ++){
  PORTC = B00000010; // RD low, WR high, CS low
  PORTC = B00000110; // RD high, WR high, CS low
    // wait for econ to go high
    while ((PINC & econ) == 0 );
  bufa[il]=PINA;
  }
endtime=micros();
sei();
}

closed 2015-09-14 by drp0

matthijskooijman, 2015-09-17:
> The issue lies with the global declaration of the unsigned int i.

I suspect that with a global variable for i, its value is forced into memory, while a local variable can be put in a register instead. Perhaps the older compiler optimized more aggressively by keeping the variable in a register during the loop and only writing it near the end, but the newer compiler is more conservative? Dunno really, but good that you've found a fix!