arduino/Arduino #3660: Need to close and open Aduino on Linux Ubuntu 15.04

34626, 2015-08-11:
Im visually impaired and have to edit my arduino code in gedit. When i wanna run it in Arduino im forced to exit the current Arduino window and open a new (CTRL + O) and select the .ino file and then i can upload it to the arduino.. It would be nice with a "reload page" shortcut.. or if you replace the file on its directory you dont need to close arduino but simple can reload page.. (Last thing is just another benefit)

BrentWilkins, 2015-08-11:
Under preferences chose 'Use external Editor'.  Your changes will be read on compile.

34626, 2015-08-11:
Very nice, but a key shortcut would be nice to it :-)

matthijskooijman, 2015-08-11:
I don't think there is a shortcut, but you only have to enable this once and it will be remembered. After that, you can use the regular ctrl-r to verify and ctrl-u to upload.

Let us know if this answers your needs sufficiently.

ffissore, 2015-09-08:
It looks like that solved @34626 issue

closed 2015-09-08 by ffissore